micropython/micropython #3341: Getting hostname with wifi station inactive causes reboot (ESP8266)

cv007, 2017-10-01:
#======================================================
# problem- getting hostname with wifi station inactive
# causes a reboot
#======================================================
#easy test
from network import WLAN,STA_IF
w = WLAN(STA_IF)
w.active(False)
w.config('dhcp_hostname')
CAUSES REBOOT

#======================================================
# I suspect wifi_station_get_hostname() is returning null when not active
# which causes strlen to start accessing address 0 (I assume bad, and
# is the cause of reset)
#======================================================

#======================================================
# possible solution- if returned char* is null, just return none type
#======================================================
# ESP8266/modnetwork.c
case QS(MP_QSTR_dhcp_hostname): {
    req_if = STATION_IF;
    char* s = wifi_station_get_hostname();   <------if s==0
    val = mp_obj_new_str(s, strlen(s), false); <-----strlen(0)

case QS(MP_QSTR_dhcp_hostname): {
    req_if = STATION_IF;
    char* s = wifi_station_get_hostname();
    val = s ? mp_obj_new_str(s, strlen(s), false) : mp_const_none;  <-----changed

dpgeorge, 2017-10-03:
I can confirm the bug.  Indeed `wifi_station_get_hostname()` returns NULL when the STA is inactive.  It's probably better to raise an exception in this case, rather than return the wrong value.  When STA is inactive you can't set the dhcp hostname either, and it raises an exception.

cv007, 2017-10-04:
I'm probably not one to contribute code, but I'll try to contribute something. 
Adding a return type of bool to req_if is done so it can be used without duplicating its code. Since 'post' error checking cannot occur after get_hostname(), we need to check the interface type before its called (req_if). Instead of also checking the active status, I am simply determining that a null return will also mean station not active (and if it somehow does return a null when active, then the 'STA inactive'  error will be inaccurate but will error nonetheless).
```
/*
WLAN(interface).config('dhcp_hostname')
arg[0]==AP_IF--------------> STA required error
arg[0]==STA_IF && active---> OK
arg[0]==STA_IF && inactive-> STA not active error
*/

//changed return type from void to bool
//so can use this function without code duplication

STATIC bool require_if(mp_obj_t wlan_if, int if_no) {
    wlan_if_obj_t *self = MP_OBJ_TO_PTR(wlan_if);
    if (self->if_id != if_no) {
        error_check(false, if_no == STATION_IF ? "STA required" : "AP required");
        return false;
    }
    return true;
}

//set val to none, in case does not get set
//if not station, 'STA required' error, break
//(val still none, req_if remains -1 so no duplicated check later)
//must be station
//just try to get hostname, whether active/inactive
//(will assume null==inactive, not null==active)
//test return value, if null- 'STA inactive' error
//if not null, set val to hostname
case QS(MP_QSTR_dhcp_hostname): {
    val = mp_const_none;
    if (!require_if(args[0], STATION_IF)) {
        break;
    }
    char* s = wifi_station_get_hostname();
    error_check(s, "STA inactive");
    if (s) {
        val = mp_obj_new_str(s, strlen(s), false);
    }
    break;
}

cv007, 2017-10-04:
I'm not a great coder, but will attempt to suggest a(nother) possible fix. The root problem I think is that these config sdk calls should not take place if the station or ap is inactive.
```
https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf
3.5. Wi-Fi Realted APIs
wifi_station_xxx APIs and other APIs which set/get configurations of the ESP8266
Station can only be called if the ESP8266 Station is enabled.

wifi_softap_xxx APIs and other APIs which set/get configurations of the ESP8266 SoftAP
can only be called if the ESP8266 SoftAP is enabled.
```
so, maybe take care of this early in the function by checking if the interface is active, but this would leave the get/set_mac_address from being able to be get/set when inactive (these appear to be the only 2 cases in this function which do not use the sdk wifi_station_xxx/wifi_softap_xxx type functions ).

```
//modnetwork.c
//line 276
STATIC mp_obj_t esp_config(size_t n_args, const mp_obj_t *args, mp_map_t *kwargs) {
    if (n_args != 1 && kwargs->used != 0) {
        mp_raise_TypeError("either pos or kw args are allowed");
    }

    //inserted code-
    if (!mp_obj_is_true(esp_active(0, args))) {
        error_check(false, self->if_id == STATION_IF ? "STA inactive" : "AP inactive" );
        return mp_const_none;
    }
```
This still requires my possible fix code in the first post, as there is otherwise nothing preventing the get_hostname from being called when AP_IF is inadvertently used, as the interface error checking is done after. The strlen(null) will be avoided, but the later require_if check will fail (and get 'STA required' error).

While I'm at it :) ... here is something else in modnetwork.c that may need attention-

```
test
>>>from network import WLAN
>>>w=WLAN(2)
>>>w.active()
no crash, will get TypeError (lucky, I guess)
>>>w=WLAN(3)
>>>w.active()
REBOOT
```
I think the cause is-
```
//modnetwork.c
//line 53
STATIC const wlan_if_obj_t wlan_objs[] = {
    {{&wlan_if_type}, STATION_IF},
    {{&wlan_if_type}, SOFTAP_IF},
};

//modnetwork.c
//line 65
STATIC mp_obj_t get_wlan(mp_uint_t n_args, const mp_obj_t *args) {
    int idx = 0;
    if (n_args > 0) {
        idx = mp_obj_get_int(args[0]);
    }
    return MP_OBJ_FROM_PTR(&wlan_objs[idx]);
}
```
as can be seen, idx can exceed the length of wlan_objs

possible fix-
```
STATIC mp_obj_t get_wlan(mp_uint_t n_args, const mp_obj_t *args) {
    int idx = 0;
    if (n_args > 0) {
        idx = mp_obj_get_int(args[0]);
    }
    if (idx >= sizeof(wlan_objs)) {
        error_check(false, "invalid WLAN interface" );
        return mp_const_none;
    }
    return MP_OBJ_FROM_PTR(&wlan_objs[idx]);
}
```

I'm exceeding my capabilities, but I hope this will help.

dpgeorge, 2018-05-17:
This was fixed in f8a5cd24d898f06f742839e0629016e704affadc by checking for a NULL pointer and in that case returning an empty string.  I tested all other config accessors and they seem to be OK when the interface is not active.

closed 2018-05-17 by dpgeorge