micropython/micropython #3342: Webrepl protocol crashes on unicode

rikte88, 2017-10-02:
Try in webrepl
>>> print('\u0414')
Disconnected

The returned data is not a valid websocket text frame, but sent as such. This causes google chrome to disconnect. Maybe endoce terminal I/O as json, or send as binary data instead of text frames?

dpgeorge, 2017-10-03:
Thanks for the report, I confirm that it's a problem (tested with Firefox).

WebSocket text frames allow data to be utf-8 encoded, so that part is OK.  The issue is that the uPy code currently sends one byte at a time over the WebSocket and so splits utf-8 data into 1-byte frames.  This is then rejected by the browser.

It might be possible to fix it by using the fragment feature of WebSocket frames to say that the rest of the utf-8 data is coming in later frames.  Or it could be fixed by sending the whole utf-8 character at once.

dpgeorge, 2017-10-03:
> Or it could be fixed by sending the whole utf-8 character at once.

See #3343 for such a fix.

dpgeorge, 2017-10-03:
The fix for this was merged.

closed 2017-10-03 by dpgeorge