micropython/micropython #3345: POLLPRI is not included in uselect

dlech, 2017-10-03:
I'm using the `poll` syscall on Linux to poll some sysfs attributes. All sysfs attributes only support `POLLPRI`, which is missing in the MicroPython implementation.

I've been adding `POLLPRI = 2` to all files using it as a workaround.

Would it make since to add `POLLPRI` to `uselect`? I'm guessing that it was omitted because it is not supported on other platforms. Another option would be to add it to `micropython-lib` (assuming that the value is standard on all unix platforms). For complete CPython compatibility, a wrapper would need to be made around the poll.register() function to change the default value for `eventmask` to `POLLIN | POLLPRI | POLLOUT` instead of `POLLIN | POLLOUT`.

Reference: https://docs.python.org/3.4/library/select.html#poll-objects

pfalcon, 2017-10-03:
Well, I'd say yeah, POLLPRI is not part of **u**select interface. But we can add it to select module in micropython-lib.

reidwagner, 2017-10-06:
So this would be a simple select module attribute POLLPRI = 0x002, and using it would look like:
`
import uselect
import select

p = uselect.poll()
p.register(0, uselect.POLLIN | uselect.POLLOUT | select.PULLPRI)
`

Am I understanding the intended library structure correctly? I'm was a little confused by this [uselect documentation](http://docs.micropython.org/en/latest/wipy/library/uselect.html) because the poll.register example uses select instead of uselect to reference  PULLIN, etc.

reidwagner, 2017-10-06:
So this would be a simple select module attribute POLLPRI = 0x002, and using it would look like:
```
import uselect
import select

p = uselect.poll()
p.register(0, uselect.POLLIN | uselect.POLLOUT | select.PULLPRI)
```
Am I understanding the intended library structure correctly? I was a little confused by this uselect documentation because the poll.register example uses select instead of uselect to reference PULLIN, etc.

pfalcon, 2017-10-06:
The above usage is possible, by definitely not recommended. There're 2 basic choices:

1. Use of MicroPython API ("u" modules), which is generally subset of Python API (maybe with some extensions).
2. Use of Python API (normal modules). Caveat: that first need to be implemented. While micropython-lib went a long way on that road for 3 years, someone looking around will find a lot of things not implemented. patches welcome.

Mixing 2 approaches is possible, but again, not recommended, done by user's at their own risk based on their understanding what they're doing ;-).

reidwagner, 2017-10-06:
Okay, got it. So the idea is just that once a functional poll API within micropython-lib's select is created, it should include POLLPRI (because that's part of CPython), and presumably dlech would bypass uselect altogether.

I'll take a look at what's involved in filling out select.py. Thanks.

pfalcon, 2017-10-06:
For starters, it should be as simple as https://github.com/micropython/micropython-lib/commit/f5bc4d8ea3a0b9fd54c7754867829b5c2516dd0c . I'm closing this issue, as `uselect` keeps not having POLLPRI, as expected, and `select` now has poll() and POLLPRI as expected too.

closed 2017-10-06 by pfalcon

dlech, 2017-10-07:
It's actually not quite that simple if you want micropython-lib to work the same as CPython. It will be a couple weeks until I can get around to doing a PR though.

pfalcon, 2017-10-07:
> It's actually not quite that simple if you want micropython-lib to work the same as CPython.

Yes, very much likely. But support for as much as possible is an idealistic goal of micropython-lib, not attainable, because MicroPython is itself not CPython. There's a trade-off between being order(s) of magnitude smaller than CPython and supporting all of it, and the emphasis is on the former, given that there's already a thing which supports all of CPython - that's CPython itself. That's in more detail discussed in https://github.com/micropython/micropython/wiki/ContributorGuidelines .

PRs are definitely welcome, but I'd suggest going thru a discussion first what doesn't work as expected and plan what to do about it. (Likely, within bounds of micropython-lib project.)

pfalcon, 2017-10-07:
Oh, if you mean the default flag value to poll() which you mentioned before, well, the obvious solution is to specify it explicit - such code will work both on uPy and CPy.

Or we can fix it - and after that, only another thousand of such fixes will remain. Again, the doc above suggests an approach towards that ("Django criteria").

dlech, 2017-10-09:
The other gotcha that got me is that you have to explicitly call File.fileno() instead of just passing a file object.