micropython/micropython #3346: Prepare for release v1.9.3

dpgeorge, 2017-10-04:
Since v1.9.2 the major change was a move to the "ports/" directory layout and that seems to be OK, so would be good to tag a new version.

Things to potentially resolve before tagging (this list may be edited):
- [ ] #3122 alloca/VLA issue
- [ ] #3291  #3344 finish cleaning up opcode enums
- [ ] #3331 math domain checking
- [ ] #3334 null bytes

robert-hh, 2017-10-04:
+1 on #3122 I understand it makes the code more robust
no opinion on finish cleaning up opcode enums
+1 on #3331 Even if no one complained about it so far. It's just consistent

dmazzella, 2017-10-12:
> plans to implement what is written in the uos.dupterm TODO?
// TODO should accept any object with read / write methods.

@dpgeorge Yes, that will be worked on soon.

News about this? Can be this released in 1.9.3?

dpgeorge, 2017-10-19:
@dmazzella In commit d6bf3658f44e22f06e0d293b8fe0542b57538b4f I made os.dupterm() on pyboard allow to take any object with write/readinto methods.  pyb.repl_uart() still behaves as before so if you want the new functionality please use os.dupterm(obj).

dmazzella, 2017-10-19:
@dpgeorge thank you, try it as soon as possible.

paganopaganino, 2017-10-22:
@dpgeorge where i can find example of use for os.dupterm?

dpgeorge, 2017-10-23:
Ok, this release was delayed a bit.  I wanted to look properly into #3122 (VLA stuff) and will attempt to do that in the coming days.  But if it's looking like it will take too long then I'll move to release v1.9.3 ASAP and postpone #3122 until after.

chrismas9, 2017-10-23:
@dpgeorge, I have been slowed down by a knee replacement. I should be back on #3361 this week.

dpgeorge, 2017-10-31:
@chrismas9 at this point I'm inclined to revert 3289b9b7a76a1230b6bb631e191a47bfc6c7a8ee so that I can do the v1.9.3 release and that it doesn't have any regressions.  Then after that the issue of compiling under Windows can be revisited.

chrismas9, 2017-10-31:
@dpgeorge, that's fine. I'm back at work now, but it's probably best to test it thoroughly without time pressure.

dpgeorge, 2017-10-31:
Thanks @chrismas9 , I've now reverted it.

The outstanding VLA issue requires some deeper changes to do it correctly so I'll postpone that until after v1.9.3.

I will tag and release v1.9.3 within the next 12 hours or so, should be by UTC 23:59, 31st Oct 2017.

dpgeorge, 2017-11-01:
Ok, v1.9.3 was tagged in fe45d78b1edd6d2202c3544797885cb0b12d4f03, see release at https://github.com/micropython/micropython/releases/tag/v1.9.3

The esp-extra-scripts branch was rebased on top of v1.9.3 and force-pushed.

closed 2017-11-01 by dpgeorge