micropython/micropython #3352: usocket getaddrinfo not working

philschenker, 2017-10-06:
I tried the module umqtt.MQTTClient and tried to add a server this didn't work. A quick debug led me to the usocket module i suppose is in this repo? I'm not sure.

This doesn't work:
import usocket
usocket.getaddrinfo("google.com", 80)

result:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -2

used micropython-image: esp8266-20170823-v1.9.2.bin

Can somebody confirm this error?

cv007, 2017-10-06:
```
MicroPython v1.9.2-8-gbf8f45cf on 2017-08-23; ESP module with ESP8266
Type "help()" for more information.
>>> from network import WLAN
>>> WLAN().isconnected()
True
>>> import usocket
>>> usocket.getaddrinfo('google.com',80)
[(2, 1, 0, '', ('172.217.6.110', 80))]
>>>
>>> WLAN().active(False)
>>> usocket.getaddrinfo('google.com',80)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -2
```
no internet- no dns lookup- error -2 from result of sdk call
here is the source code for the callback result from the sdk call-
extmod/modlwip.c-
```
// Callback for incoming DNS requests.
STATIC void lwip_getaddrinfo_cb(const char *name, ip_addr_t *ipaddr, void *arg) {
    getaddrinfo_state_t *state = arg;
    if (ipaddr != NULL) {
        state->status = 1;
        state->ipaddr = *ipaddr;
    } else {
        // error
        state->status = -2;
    }
}
```

philschenker, 2017-10-06:
Thanks cv007 you got me to the right point, now it works.
Of course the internet connection was one of the first things I checked. I even put in the IP address instead of domain and it worked. I have no idea what went wrong, maybe the internet connection of my phone?

anyway. This is not a bug.

closed 2017-10-06 by philschenker