micropython/micropython #3356: sys.stdin does not handle ctrl-D EOF markers

dhalbert, 2017-10-07:
`sys.stdin.read()`, `sys.stdin.readlines()`, etc. do not handle ctrl-D at the beginning of a line as EOF (except on ports like unix). So for instance, there's no way to terminate input when doing `sys.stdin.readlines()`.

Is this something that should mimic CPython on an OS-less port? Some lower-level stream read routine could check if the stream was `sys.stdin` and handle ctrl_D specially.

robert-hh, 2017-10-08:
I had to learn too that sys.stdin(), sys.stdout() are not implemented as a stream. That could indeed be helpful. But then a method for setting it transparent would also be needed for cases, where Ctrl-D shall not be seen as special.
You could use a crutch, and set Ctrl-D as the interrupt char with micropython.kbd_intr(4) and catch the keyboard interrupt exception, which happens then. That however triggers on every Ctrl-D, not only on a Ctrl-D at the beginning of a line. But sys.stdin() isn't transparent anyhow.
Alternatively you can use readline() and check for a line consisting of Ctr-D\n.

robert-hh, 2017-10-08:
Yes, there are cases when sys.stdin() and sys.stdout() being a stream would be useful. In your case, you could make your own readlines() function, like:
```
import sys
def readlines():
    lines = []
    while True:
        line = sys.stdin.readline()
        if line != '\x04\n':
            lines.append(line)
        else:
            break
    return lines
```

dpgeorge, 2017-10-10:
The problem is that bare-metal ports don't really have a notion of "end of file" when talking about the incoming REPL stream.  On unix-like systems the input is "cooked" and shell/OS determines what an end of file is: pressing ctrl-D on an empty line will do it but uPy never sees that ctrl-D character, it just sees that there are no more bytes to read.

But probably it makes sense to interpret a ctrl-D character as meaning end-of-file on bare-metal, otherwise sys.stdin.readlines() is useless.  Note that this processing would be special to the sys.stdin reading function (which already has special handling for newline).  For raw input there's still sys.stdin.buffer.

dhylands, 2017-10-10:
If we add support for Control-D it would be good to also support Control-C. Currently there is no Control-C handling on serial REPLs (at least on stm32).

dpgeorge, 2017-10-10:
> it would be good to also support Control-C

Yes, agreed.  But that's at a lower-level than sys.stdin because ctrl-C handling needs to be asynchronous (eg handled from an interrupt).