micropython/micropython #3357: mpy-cross compiled code behaves incorrectly

NathanWilliams, 2017-10-08:
I'm working on an esp8266 where ram is tight.
I am using mpy-cross to skip the onboard compilation step, which helps a lot for ram.

However I have found a case where the compiled behaviour is incorrect compared to non-compiled code.

This is the code:
`def x(n=8):
    return [0]*n`

On the REPL or a non-compiled .py file, you get back:
`[0,0,0,0,0,0,0,0]`

But from the .mpy file, you get back: True
(Or in some other cases False).

As a workaround, I am now doing something like:

`def x(n=8):
    return eval('[0]*{}'.format(n))`

It isn't pretty, but lets me keep using mpy-cross.

NathanWilliams, 2017-10-08:
I have found another case of incorrect behaviour.

`def t5():
    for i in range(0,16):
        print(i)`

Normal prints 0 to 15
cross compiled prints 0 in an infinite loop.

I think there is also a problem with list addition, again resulting in a boolean.
I haven't been able to make a test case that shows the problem yet, but I know I solved some incorrect code by switching to using extend on the lists.

peterhinch, 2017-10-08:
I can't replicate this using today's source on ESP8266. These problems usually arise when an out of date cross-compiler is used. Have you recompiled it?
```
MicroPython v1.9.2-152-gc15be98 on 2017-10-08; ESP module with ESP8266
```

NathanWilliams, 2017-10-08:
That sounds reasonable.

I used the officially released firmware, but compiled may-cross from HEAD.
I'll build them from the same commit and get back.

Thanks for the help, I was pulling my hair out!

peterhinch, 2017-10-09:
@dpgeorge This issue occurs regularly. Is there any scope for ensuring that a sensible exception is thrown?

robert-hh, 2017-10-09:
That must have been an old firmware on the unit. The newer ones complain about an incompatible mpy-cross version. Been there a few times, for instance yesterday after last weeks bytecode change.

NathanWilliams, 2017-10-09:
I have hit problems compiling, and I am on a deadline, so I'll have to try again later.
I have a workaround that is working for me right now.

For reference, I flashed on the latest available from the website: esp8266-20170823-v1.9.2.bin
I did try a daily build, and I got the error @robert-hh mentioned.

I don't want to waste anymore of your time, so I'll just close this issue as it seems to be my problem, not the code.

As an aside, could mpy-cross possibly be put up for download, it is a useful tool, even if you don't intend on building from source.

closed 2017-10-09 by NathanWilliams

robert-hh, 2017-10-09:
The latest build available on the web site is the daily build esp8266-20171009-v1.9.2-153-g53966fd9.bin
That one should work in combinatikon with a fresh built mpy-corss. There was a bytecode change last week. And I agree to put mpy-cross on the download page, at least for a few major OS'es, like Ubuntu/Debian, OS X and Windows.
And I do not agree that it is your problem, if people get lost in a version jungle.

dpgeorge, 2017-10-09:
Yes there was recently a change is the bytecode.  Usually the .mpy version number would be updated straightaway but there was a bit of a delay and so there were a few versions where you could run incompatible bytecode without getting an error.

Re mpy-cross on the downloads page: it would be nice to do, and the simplest thing would be a statically compiled version for *nix, plus versions for Mac and Windows.  But until that happens please just build from this repo.