micropython/micropython #3361: micropython.exe: Building with MINGW fails

robert-hh, 2017-10-09:
The commit 3289b9b added $(PROG_EXT) to the target names causes the windows build of micropython.exe to fail with MINGW, because then the filename is micropython.exe.exe. 
Cure: change in ports/windows/Makefile:
PROG = micropython

chrismas9, 2017-10-10:
I'm working on it. The fix is a bit more complicated because changing to `PROG = micropython` breaks cross compiling for Windows under Linux. Setting `PROG_EXT` needs to be based on the target environment, not the host environment.

If WIndows port can be cross compiled under Linux then a solution should allow `mpy-cross.exe` to be cross complied under Linux as well

chrismas9, 2017-10-11:
@robert-hh Do you get 
```LINK micropython.exe
make: i686-w64-mingw32-strip: Command not found
make: *** [../../py/mkrules.mk:133: micropython.exe] Error 127
make: *** Deleting file 'micropython.exe'
```
I had been using MSYS1 to build mpy-cross.exe and have just installed MSYS2 and MINGW64 to build Windows port. For native compile binutils (strip, size, etc) should not be prefixed with `i686-w64-mingw32-` so mkenv.mk needs a bit more work to fix Windows builds properly.

@dpgeorge Since this issue was raised I have been reviewing all the Windows native and cross compile make processes and have found a few more issues that need small fixes.

What is your timeframe for **v1.9.3**? I have worked out a solution, but there's quite a lot of testing because everything done for Windows port affects mpy-cross, and changes for native compile can break cross compile.

What are the rules for including `fmode.c`? At the moment `mpy-cross.exe` compiles without `fmode.c` if you use MSYS1 (includes `gcc)` or MSYS2 (with `gcc` installed). `fmode.c` is included if you compile with MINGW (using `i686-w64-mingw32-gcc)`.

STM32 port auto builds `mpy-cross` if missing but only if `gcc` is present (from MSYS). There are probably quite a few Windows users out there like me that had MYS1 installed and found that `mpy-cross` built ok. I don't think they should be forced to install MINGW as well. Hence my question about `fmode.c`. The rule for including it may need to be changed depending on which compiler is used.

Do you want to review the fixes incrementally?

robert-hh, 2017-10-11:
@chrismas9 I got your mail, but no text here.
> make: i686-w64-mingw32-strip: Command not found
> make: *** [../../py/mkrules.mk:133: micropython.exe] Error 127
> make: *** Deleting file 'micropython.exe'
Yes, I see that, when building mpy-cross, because it looks for mpy-cross, not mpy-cross.exe, and that both causes the error message witjh the strip command and prevents deleting of mpy-cross.exe. So I have an error, which I happily can ignore, and I'm happy with what I got.

stinos, 2017-10-12:
> What are the rules for including fmode.c? At the moment mpy-cross.exe compiles without fmode.c if you use MSYS1 (includes gcc) or MSYS2 (with gcc installed). fmode.c is included if you compile with MINGW (using i686-w64-mingw32-gcc).

That was done on purpose, see https://github.com/micropython/micropython/commit/5b57ae985ff7064dd7b09b0ce891697bfaa5dae2

>  For native compile binutils (strip, size, etc) should not be prefixed with i686-w64-mingw32-

Why would you use strip if i686-w64-mingw32-strip exists and you're using i686-w64-mingw32-gcc and the likes?

chrismas9, 2017-10-12:
@stinos Thanks for the pointer about fmode.c. Makes sense.

> Why would you use strip if i686-w64-mingw32-strip exists and you're using i686-w64-mingw32-gcc and the likes?

Because `i686-w64-mingw32-strip` does not exist for native builds. See explanation here:

https://sourceforge.net/p/mingw-w64/mailman/mingw-w64-public/?viewmonth=201203&viewday=24

>  Should it be so that all executables files in <prefix>/bin have the  prefix i686-w64-mingw32-*?

>If and only if you are building a cross-compiler (i686 to x86_64 Windows counts as cross-compiler). Then everything will be prefixed and nothing will be prefixless.

>If you are building a native toolchain, only four or five gcc executables (not counting the new 4.7 plugin wrappers) are prefixed.

>Binutils are not prefixed if built as non-cross.

As far as I can see the solution is to create a new variable CROSS_UTILS which will be set equal to CROSS_COMPILE unless CROSS_COMPILE contains `mingw` and OS = Windows_NT. This will be automatic with no user facing changes.

How has this worked in the past? Is anyone building the Windows port with native mingw, or only cross mingw on Linux or MSVC?

stinos, 2017-10-12:
> native builds

What exactly do you mean with that?

> How has this worked in the past?

When I test something using mingw-w64 I always did this on cygwin (since I happened to have that anyway). Which worked well until your commit. And there's a cross-compile step in the Travis build as well which always worked ok.
I did once test using mingw-w64 on MSYS2 if I recall correctly (might be wrong, not sure, too long ago) but definitely had no build problems related to strip etc.
But note there is this: https://github.com/micropython/micropython/issues/1751

chrismas9, 2017-10-12:
> native builds

https://stackoverflow.com/questions/771756/what-is-the-difference-between-cygwin-and-mingw

> Cygwin is an attempt to create  a complete UNIX/POSIX environment on Windows. To do this it uses various DLLs. While these DLLs are covered by GPLv3+, their license contains an exception that does not force a derived work to be covered by the GPLv3+. MinGW is a C/C++ compiler suite which allows you to create Windows executables without dependency on such DLLs - you only need the normal MSVC runtimes, which are part of any normal Microsoft Windows installation.

When you build under Cygwin you are building within a POSIX environment with Linux tools. The gcc toolchain that generates a WIndows stand alone .exe file is a cross compiler so it has  i686-w64-mingw32-strip.

I am trying to create the simplest Windows environment for people who only want to freeze modules or change build options. That means not installing Cygwin and not having to work in a POSIX environment under the Cygwin shell. The simplest solution is to just install MSYS2 and a couple of packages with pacman. After that you can build at an MS command prompt or use Windows batch files or links.

When MSYS2 / MINGW64 are installed you get i686-w64-mingw32-gcc.exe and strip.exe. These are native Windows programs. Using native tools on the same host as the target (micropython.exe) is what I mean by native builds.

@stinos I am sorry I broke the cross compile process. I will see this through and fix and test it for all platforms. Discussing it with you has been very useful and I have a better insight into how people are using Windows. I need to setup 5 test environments 1. MSYS / MINGW 2. Cygwin 3. WSL 4. Linux. 5. Plain Windows. The latter being for stand-alone testing of micropython.exe.

> I did once test using mingw-w64 on MSYS2 if I recall correctly (might be wrong, not sure, too long ago) but definitely had no build problems related to strip etc.

Was Cygwin still installed? make would have found `i686-w64-mingw32-strip` on the Cygwin path even though it would have been missing on the mingw path.

Once I finish this I will look at #1751

stinos, 2017-10-12:
> When MSYS2 / MINGW64 are installed you get i686-w64-mingw32-gcc.exe and strip.exe

Yes, but then you start via mingw32_shell.bat for example, and then you also get gcc.exe (which as far as I know is the same one as i686-w64-mingw32-gcc.exe) in your PATH, so you just call ```make``` without specifying CROSS_COMPILE and that's it. That is likely why I never had problems with missing i686-w64-mingw32-strip.exe.

> make would have found i686-w64-mingw32-strip on the Cygwin path even though it would have been missing on the mingw path.

That would only be so if the Cygwin path is known to make, which isn't the case when you start make from an MSYS2 shell. Maybe if you'd start the MSYS2 shell from within the Cygwin shell or so, but that would be madness.

robert-hh, 2017-10-12:
I'm working with MINGW32 a while building stand-alone Windows binaries fro mpy-cross and MicroPython. At mpy-cross, ther was always the error, that a commans `strip mpy-cross` should be executed, instead of `strip mpy-cross.exe`.  That did not harm too much, since mpy-cross.exe stayed. Make did not find mpy-cross to delete it.
Building micropython.exe worked fine, until the lastest change of the makefile, which appended .exe to the name micropython.exe, resulting in a call of `strip micropython.exe.exe`, which obviously failed too. But in this case Make deletes Micropython.exe, which is not good. So I had to change the make file in calling the target just micropython.
So things are a little bit messed up.

chrismas9, 2017-10-12:
Point taken about paths. I forgot MSYS and Cygwin don't import the Windows path.

>but then you start via mingw32_shell.bat 

The whole idea is not to do that. I am trying to set up and document a completely Windows user friendly environment. My tools of choice are UltraEdit or Notepad++, UltraCompare or WinMerge , Eclipse and TortoiseGIT. No command line, no learning cryptic commands and options that I promptly forget. I am only going through that pain now so I can create a full Windows GUI dev environment for MicroPython.

>and then you also get gcc.exe (which as far as I know is the same one as i686-w64-mingw32-gcc.exe

No, you have to install gcc.exe with pacman and it is a different beast:

>C:\Users\cmaso>gcc --version
>gcc (GCC) 6.3.0

>C:\Users\cmaso>i686-w64-mingw32-gcc --version
>i686-w64-mingw32-gcc (Rev1, Built by MSYS2 project) 7.2.0

gcc is needed to build mpy-cross.exe but won't build micropython. "make" fails using gcc:

```../../py/obj.h: In function 'MP_OBJ_IS_SMALL_INT':
../../py/obj.h:85:17: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
     { return ((((mp_int_t)(o)) & 1) != 0); }
                 ^
```
"make CROSS_COMPILE = i686-w64-mingw32-" works fine once binutils (strip.exe, etc) is fixed.

My conclusion is that you  need CROSS_COMPILE = i686-w64-mingw32-
I would like to add CROSS_COMPILE ?= i686-w64-mingw32- to the windows Makefile, then you would be able to just run "make", but still override it.

chrismas9, 2017-10-12:
>So things are a little bit messed up.

Yes, but I am doing a very thorough investigation of all the build environments and will fix it.

>I'm working with MINGW32 ... Building micropython.exe worked fine

oh dear! I will have to build an MINGW32 environment as well. I'm using MINGW64 and it fails at strip:

```LINK micropython.exe
make: i686-w64-mingw32-strip: Command not found
make: *** [../../py/mkrules.mk:133: micropython.exe] Error 127
make: *** Deleting file 'micropython.exe'
```

robert-hh, 2017-10-12:
gcc --version tells:
`gcc (i686-win32-dwarf-rev1, Built by MinGW-W64 project) 6.2.0
Copyright (C) 2016 Free Software Foundation, Inc.`

stinos, 2017-10-12:
> The whole idea is not to do that. I am trying to set up and document a completely Windows user friendly environment. The whole idea is not to do that. I am trying to set up and document a completely Windows user friendly environment. 

Well that sort of depends on your definition of user friendly I guess. From my point of view, you are just trading some simple command line steps (about 3 in my case I think: get MSYS2 base, have pacman install mingw and git, done.) for a bunch of downloads and possibly manual configuration steps via a gui, which you'd also have to remember? Or do all these installers by default behave such that Eclipse automatically finds mingw from the MSYS2 installation)? And arguable the simplest and most friendly way would be to just download VS2017, with a couple of clicks enable it to install git and C++ tools and that's it.

> No, you have to install gcc.exe with pacman and it is a different beast:

Ah but my point is you don't do that. You don't need it for uPy, you just use ```pacman -S mingw-w64-i686-gcc```  and then you get

```
# gcc --version
gcc.exe (Rev1, Built by MSYS2 project) 7.2.0
Copyright (C) 2017 Free Software Foundation, Inc.

# i686-w64-mingw32-gcc --version
i686-w64-mingw32-gcc.exe (Rev1, Built by MSYS2 project) 7.2.0
Copyright (C) 2017 Free Software Foundation, Inc.
```

and you can just ```make``` both micropython and mpy-cross. So you don't strictly need CROSS_COMPILE = i686-w64-mingw32-, if you keep it simple.

stinos, 2017-10-12:
Anyway, since there seem to be so many different ways of building with mingw this will have to be tested thoroughly for all possibilities currently known I guess. And the README should be updated to clearly state how uPy is currently supposed to be built using mingw (next to the current explanation of how to do it using Cygwin).

dpgeorge, 2017-10-13:
> What is your timeframe for v1.9.3? I have worked out a solution, but there's quite a lot of testing because ...

It's not urgent but next week (16th-30th Oct 2017) would be good.

As for the rest of the discussion here: I don't think it's feasible to support every possible combination of Windows environments, not least because it's very difficult to test them all in an automated way so that things don't break again in the future.

So can we just pick a few common ways to build under Windows / MINGW and then document and support those?

dpgeorge, 2017-10-31:
Ok, the commit 3289b9b7a76a1230b6bb631e191a47bfc6c7a8ee was reverted in 02b4b23319a84d5b96b9394d8c379ef0549fd208 until a better solution can be found.

closed 2021-03-07 by robert-hh