micropython/micropython #3364: formatting of single character codes in strings is truncated to 1 byte

goatchurchprime, 2017-10-10:
The problem is the %c or {:c} which is supposed to format a number into a unicode character, but in Micropython (tested on the esp8266 and esp32) seems to truncate the parameter at one byte.

```
MicroPython v1.9.2-276-ga9517c04 on 2017-10-09; ESP32 module with ESP32
Type "help()" for more information.
>>> print("a%cb" % (65+256))
aAb
>>> print("a{:c}b".format(65+256))
aAb
>>> print("a{:s}b".format(chr(65+256)))
aŁb
```

```
Python 3.6.1 |Anaconda custom (64-bit)| (default, May 11 2017, 13:09:58) 
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> print("a%cb" % (65+256))
aŁb
>>> print("a{:c}b".format(65+256))
aŁb
>>> print("a{:c}b".format(65))
aAb
>>> print("a{:c}".format(0x10189))
a𐆉
```

(Hm. I did not know that unicode went beyond 2 bytes.  That's got to be a hassle to implement.)

ajlennon, 2017-10-10:
My guess is this is related to use of the underlying printf() function which is probably ASCII based on the ESP8266? Sometimes it can be a build time define to switch between byte-wide ASCII and word-wide Unicode...

ref: https://github.com/micropython/micropython/issues/209

dpgeorge, 2017-10-11:
Yes this is an issue.  Unicode is not handled well in string formatting functions, for example there are related issues to width formatting, eg `print('|%4s|' % '\u0180')`  only prints 2 spaces of padding, not 3 as expected.

pfalcon, 2017-10-28:
This is prio-low, because CPython-compatible workarounds exist (use "%s", use chr(), etc.).