micropython/micropython #3366: drivers/nrf24l01/nrf24l01test.py Problems with machine.idle().

peterhinch, 2017-10-11:
There are two issues with [this line](https://github.com/micropython/micropython/blob/a3afa8cfc46913f471d5eb55da4ae22dee92c25f/drivers/nrf24l01/nrf24l01test.py#L72). Firstly the machine module is not imported so the `slave()` function throws an exception.

Once fixed the call to `idle()` causes serious performance problems. I tested using two Pyboards with radios about 3 metres apart in the same room. With `idle()` the round-trip response was on the order of 300ms (with considerable variability). With the line commented out it was a consistent 17ms, which is similar to the performance of the old `pyb` based code.

It seems that `machine.idle()` behaves differently from `pyb.wfi()`.

dhylands, 2017-10-11:
Hmm. machine.idle() calls exactly the same function as pyb.wfi():
https://github.com/micropython/micropython/blob/a3afa8cfc46913f471d5eb55da4ae22dee92c25f/ports/stm32/modmachine.c#L537
https://github.com/micropython/micropython/blob/a3afa8cfc46913f471d5eb55da4ae22dee92c25f/ports/stm32/modpyb.c#L119

peterhinch, 2017-10-12:
Thanks for that. Evidently something else has changed in the migration from `pyb` to `machine` which is causing it to behave in this way. I'll investigate further.

peterhinch, 2017-10-15:
I've posted some code [here](https://github.com/peterhinch/micropython/tree/nrf24l01_portable) for comments. The code has been tested on Pyboard, ESP8266 and ESP32. Targets were at close range within one room.

`nrf24l01test.py` contains config data for my hardware and I'll remove this before submitting a PR. Key points:
1. Platform detection at runtime with config data stored in a dict.
2. The call to `machine.idle()` is removed. It proved problematic even between Pyboards as described above.
3. The `NRF24L01` constructor has an optional `baudrate` arg to facilitate use with soft SPI.
4. nrf24l01test.slave() has an additional delay to allow the (potentially slow) remote master time to switch from transmit to receive mode. The key delays are changed to constants and commented at the start of the code as a guide to porting to further platforms.
5. Redundant code in `nrf24l01.py` removed.

SpotlightKid, 2017-11-29:
Fix is merged. This issue can be closed, right?

peterhinch, 2017-11-29:
Yes.

closed 2017-11-29 by peterhinch