micropython/micropython #3367: PYB: REPL is a little unusual.

Li-Weiwei, 2017-10-11:
After commit 980b33177bbb99f4bc6472c24a0cc67bf7760679, when I run **screen /dev/ttyACM0**, there is no previous output.  
When I press the **Enter** key, it outputs the following:
```python

Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>>
```
If I press Ctrl+D again, it outputs the following:
```python
PYB: sync filesystems
PYB: soft reboot
MicroPython v1.9.2-104-g980b331 on 2017-10-11; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>>
```
I did not find where the problem lies, what is the reason for this?

robert-hh, 2017-10-11:
It is pretty common that there are some spurious bytes in the input buffer when the USB interface is started. Just ignore that, or push Ctrl-C as the first char.

Li-Weiwei, 2017-10-11:
Thank you. Before commit 980b33177bbb99f4bc6472c24a0cc67bf7760679, it didn`t happen. Should it be fixed?

robert-hh, 2017-10-11:
I personally do not see that as an issue, more like a typical glitch when starting a communication link. Like a harrumphing at the start of a talk. Yes , one could clear the input buffer before starting the REPL That might catch a few of these cases. But unless there is a strong reason, I would not go for it.

Li-Weiwei, 2017-10-11:
Thank you very much. :blush:

dpgeorge, 2017-10-13:
@Li-Weiwei thanks for reporting this, it is indeed a regression from earlier behavior, introduced by that commit you point to.

dpgeorge, 2017-10-13:
The output buffering issues should be fixed by 7c7c7b161df5be99aee9dea444b6e4beacf78438, which means that you should now see the uPy banner when you access the REPL for the first time after a boot via USB power.

@Li-Weiwei I can't reproduce the issue with SyntaxError when first pressing enter.  Can you confirm if it's still an issue with the latest build, or not?

Li-Weiwei, 2017-10-13:
@dpgeorge I tested it on Windows7 and macOS10.12.6 it has been fixed, but there will still be problems on Ubuntu.  
Ubuntu14.04 in VirtualBox:
```
0
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> ~x~~x~
```
or
```
~x~
```
or I have to press Enter a few times to get in.  
Ubuntu 16.04 has the same phenomenon.

dpgeorge, 2017-10-13:
@Li-Weiwei thanks for testing.  Do you have the input errors back before the USB CDC changes ie does ede8a0235b47333aaaa8d03a3b9e20b014be3808 show any of these input errors?

Li-Weiwei, 2017-10-13:
@dpgeorge On Ubuntu, ede8a0235b47333aaaa8d03a3b9e20b014be3808 has these errors too.  
Are these errors on Ubuntu the system itself or any other reason?

dpgeorge, 2017-10-13:
> Are these errors on Ubuntu the system itself or any other reason?

Yes it looks like this is an Ubuntu-specific problem: the ModemManager is running in the background and automatically sending AT commands to the pyboard when it is plugged in.  This also happens for other boards (non-pyboard) that appear as /dev/ttyACM0.  To fix it you can uninstall ModemManager; see https://askubuntu.com/questions/923426/odd-characters-when-usb-serial-port-connection-made and https://unix.stackexchange.com/questions/270989/disable-automatic-serial-port-probing-in-linuxfedora

Li-Weiwei, 2017-10-13:
Yes, as Damien recommended, on Ubuntu16.04 on a physical computer, uninstall ModemManager will solve this problem.  
On Ubuntu14.04 in VirtualBox, uninstall ModemManager is useless. So I guess there may be other reasons on Ubuntu14.04(in VirtualBox).

Li-Weiwei, 2017-10-13:
Such as robert-hh recommended, I think I should let it(Ubuntu14.04 in VirtualBox) go.

dpgeorge, 2017-10-13:
> On Ubuntu14.04 in VirtualBox, uninstall ModemManager is useless.

I tried this exact thing and it worked for me.

Anyway, there's nothing more that can be done here to fix the AT commands being sent to the board (unless the board detected these and ignored them, but that's not a good idea).  So this issue can be closed.

closed 2017-10-13 by dpgeorge

robert-hh, 2017-10-13:
Besides this special topic beeing sufficiently covered, thinking about handling that technically, if at all, I got some other thougths:
As far as I recall, there is still the inititive hangign on to generalize the REPL input source & handling. Doing that, a device indepedent method of either 
a) telling, how many (or if any) characters are waiting already in the input buffer, or 
b) a method of flushing the input buffer
On some ports & board that is possible, like option a) on Pyboard with USV_VCP.any(), or for UARTs with uart.any(), but it differs too much for different interfaces and boards.

dhylands, 2017-10-18:
Creating a udev rule and adding ID_MM_DEVICE_IGNORE to the environment, as is done here:
https://github.com/micropython/micropython/wiki/Board-STM32F407-Discovery
will cause modem manager to stop sending commands to a given serial port.