micropython/micropython #3369: mpy-tool shebang enforces python3

infinnovation, 2017-10-12:
tools/mpy-tool.py was made compatible with both python 3 and python 2 in commit c3beb16db33e941021ff839a2fa9c806c134aa48, but the first line is

```sh
#!/usr/bin/env python3
```
which makes it fail on systems with no python executable called "python3".

Suggest s/python3/python/

dpgeorge, 2017-10-13:
This script is (usually) called via `$(PYTHON) $(MPY_TOOL)` in the build process, and that evaluates to `python tools/mpy-tool.py` in the default set-up, which should support Python 2.  If you execute this script yourself (ie not part of the build process) then you can use Python 2 via: `python2 tools/mpy-tool.py ...`.

But anyway, this issue is related to #1616 and a way needs to be found to support Python 2 and 3, or just Python 3 consistently with a documented way to use Python 2 if needed.

dpgeorge, 2019-06-24:
#1616 was resolved and hence this issue as well, see 5368210e36f1d05efb0ea2c09661ebc80688880b

closed 2019-06-24 by dpgeorge