micropython/micropython #3372: for i in range(7.6):  print(i) does not throw a TypeError

goatchurchprime, 2017-10-16:
If you just write:
`range(7.6)
`
On its own you get the proper response:
`TypeError: can't convert float to int
`

There may be a slight skipping of intermediate stages with the implementation of  
`for i in range(7.6):
`
which causes it to miss out this error.

robert-hh, 2017-10-16:
funny enough, the same for ... structure in:
`[_ for _ in range(7.6)]`
also raises an exception.

dpgeorge, 2017-10-17:
This is a consequence of the optimisation that the compiler does, so that for-range constructs don't allocate on the heap.  It may be possible to fix it in a clean way.

dpgeorge, 2022-06-24:
This issue can also give wrong line numbers, eg in:
```python
for i in range([1, 2]):
    1
    2
    3
```
it will report the error with the list not being an integer at line 4 (the end of the loop).

One way to fix this would be to not optimise for-range loops, and instead use the range iterator object.  This no longer requires a heap allocation, but does require 4 extra entries on the Python stack of the function.  It's not clear if this is a win.