micropython/micropython #3373: The Wiznet driver, for both 5200 and 5500 chipsets, doesn't support SO_REUSEADDR

Lisa999, 2017-10-16:
> The Wiznet driver, for both 5200 and 5500 chipsets, doesn't (yet) support SO_REUSEADDR, non-blocking mode or ioctl (for poll'ing). 
Meaning that uasyncio isn't working on a pyboard with either the W5200/W5500 ethernet shield. Strange that it's so hard for an IOT (I for Internet...) device to make an connection to the network...
Without polling, you're basically waiting for an ethernet connection, blocking the GPIO readout! ;-(

dpgeorge, 2017-10-17:
As a start, Wiznet ioctl for poll read and write was implemented in 8fa3d2996c9d69b1a6ba20f6beb354f4ce0001c8

Lisa999, 2017-10-17:
>>> dir(usocket.socket)
>>> ['__del__', 'close', 'bind', 'listen', 'accept', 'connect', 'send', 'recv', 'sendto', 'recvfrom', 'setsockopt', 'settimeout', 'setblocking']
No read/readline/write attributes??? Guess nobody is using the W5500/5200 with uasyncio...

dpgeorge, 2017-10-20:
Full support for Wiznet is provided by PR #3379.

closed 2017-11-08 by Lisa999

dpgeorge, 2022-07-15:
@UncleBee1885 if you have an issue, please open a new bug with a full bug report (hardware and firmware used, script run, exact steps to reproduce).