micropython/micropython #3374: Txpower in network scan

pedrofelipest, 2017-10-16:
Would you like to return from network.scan() the txpower information?

dpgeorge, 2017-10-17:
> Would you like to return from network.scan() the txpower information?

I'm not quite sure what you mean... network.scan() results already return an RSSI value for each AP.  If the STA is doing an active scan then the tx-power used in the scan would be better returned via the network.status() method (see #3351), but I doubt that the wifi driver even exposes that value.

pedrofelipest, 2017-10-17:
I want to receive the value of AP txpower next to RSSI to calculate a distance between the devices. You see?

dpgeorge, 2017-10-17:
> I want to receive the value of AP txpower next to RSSI to calculate a distance between the devices. You see?

I don't think the beacon frames from the AP have the TX power encdode in them.  And even if they did it wouldn't be very useful information because these values are hard to calibrate.  Even the RSSI value can be dependent on the particular wifi chip.

To do what you want to do you need to have a very controlled environment with known transmit power from the APs.  See eg https://stackoverflow.com/questions/11217674/how-to-calculate-distance-from-wifi-router-using-signal-strength

tuupola, 2017-10-18:
Distance calculation from RSSI is just an approximation. However if you are trying to figure out location and you have three or more access points you can get good results with Nonlinear Least Squares fitting. See [Trilateration With n Points](https://appelsiini.net/2017/trilateration-with-n-points/) in the end of this article for an example.

peterhinch, 2017-10-18:
Trilateration is an algorithm which does not require knowledge of TX power. Calculation of distance from TX power and RSSI is subject to so many variables that I doubt it could ever be remotely accurate. Propagation conditions at 2.4GHz are highly variable. Antennas vary, some AP's use beamforming, polarisation may be an issue...

virtualguy, 2017-10-18:
I totally agree that rssi based location isn't great but Google seems to
have made really good use of it for improving location services
particularly for maps. Obviously the resolution isnt down to 2m but it does
provide some location information. Also if you actually have controlled
wifi infrastructure then results can certainly be improved. Cisco has a
location system based on WiFi rssi though I believe it runs on the AP's
rssi.

In any case it really depends on the application and what level of accuracy
you need.

It would also be great to know the rssi for iot devices in fixed locations
to diagnose intermittent connectivity issues.

dpgeorge, 2017-10-19:
The main point in regard to the original issue is that it's not possible to get tx-power from the AP beacon frames.  So closing the issue.

closed 2017-10-19 by dpgeorge