micropython/micropython #3376: unix: dir() of a superclass instance does not show subclass methods

peterhinch, 2017-10-18:
This behaviour differs from CPython. Apologies if this is by design:
```python
class FooSub():
    def bar(self):
        pass
 
class Foo(FooSub):
    def rats():
        pass
 
foo = Foo()
dir(foo)
```
Outcome:
```
['__module__', 'rats', '__qualname__']
```
Tab completion also fails to access `bar()`.

dpgeorge, 2017-10-18:
Yes, this is an unimplemented feature, and I recently had another request for it via a different channel.  It's tricky to do in a compact/efficient way, not least because of the possibility of multiple inheritance.  One idea I had was to simply iterate over all qstrs and use `mp_load_method_maybe()` to check if that qstr is available in the object.  That would be shorter code than currently exists and would cover almost all cases.  But it will probably be terribly inefficient due to the 1000's of lookups needed.

pfalcon, 2017-10-19:
I figure there's a mix-up in the title.

peterhinch, 2017-10-20:
Oops!

dpgeorge, 2017-11-22:
> One idea I had was to simply iterate over all qstrs and use mp_load_method_maybe() to check if that qstr is available in the object.

I had a go at implementing this idea to fix both dir() and tab-autocompletion.  It's simple and works very well, and saves quite a few bytes of code, around 200-300 bytes.  But, as expected, it's a lot slower.  A dir() or tab-autocompletion is roughly 150 times slower than before.  In absolute time it goes from about 100us to 15ms (very roughly).

Now, dir() is mostly used in test suites (to get the list of tests to run) and for user introspection.  It's not usually part of time-critical code.  And tab-autocompletion is also purely a user-interaction thing so not that time critical, as long as the delay is not so noticeable to the user.  But it is noticeable on slow MCUs: the microbit runs at 16MHz and using the qstr iteration makes pressing "tab" at a bare prompt quite slow, around 150-200ms slow.  Compare that to before when it felt almost instant when pressing "tab".

So the options for this ticket seem to be:
1. Make no change and don't fix the problem.  dir() and tab-autocomplete only list the top-most entities in an instance (nothing from a parent class).
2. Iterate through all qstrs to find valid entries.  This makes dir() and tab-autocomplete list pretty much everything available in the object, including all parents, including multiple inheritance, and being sure not to double count things if a base class appears more than once in the inheritance tree (which is otherwise non-trivial to do).  Pros: makes functionality complete, reduces code size; cons: very slow.
3. Find some middle-ground implementation that does a simple search through parent classes, most likely ignoring multiple inheritance because that will really blow out the implementation of autocomplete.

peterhinch, 2017-11-22:
I think it's worth fixing. `dir()` is particularly useful for beginners who may lack skill for studying the source. Hence it would benefit many users of the MicroBit. By the same token it befits an educational platform to produce correct results. I'd have thought that tab completion in 200ms would be acceptable, but without experiencing it in practice I guess it's hard to be dogmatic about it.

PR #3617 (2018-02-19): Improve dir() and REPL autocomplete by using "qstr probing" to find names
This set of patches improves builtin dir() and REPL autocomplete by "probing" the target object with all qstrs in order to do the search for valid attributes for that object.  Basically it goes through all known names and checks if that name is an attribute of the object.

This is of course slower than the previous implementation which just iterated through the relevant dictionaries, but it allows for a much more complete search of names.  In particular this new algorithm allows to list all attributes of a user-defined c [... cut]
closed 2018-02-19 by dpgeorge