micropython/micropython #3384: drivers/display/ssd1306 SPI driver fails to import

zgyarmati, 2017-10-22:
On STM32F4DISCOVERY the SSD1306 display driver's SPI implementation fails to get imported with the following error:
```
>>> d = SSD1306_SPI(width, height, spibus, dc, res, cs)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ssd1306.py", line 144, in __init__
  File "ssd1306.py", line 49, in __init__
  File "ssd1306.py", line 74, in init_display
  File "ssd1306.py", line 147, in write_cmd
TypeError: 'mode' argument required

```

Adding the missing  mode parameter fixes the issue.  PR will follow soon.

dpgeorge, 2017-10-22:
The ssd1306.py driver was designed to work with the machine.SPI class, and it looks like you are using the legacy pyb.SPI class instead.  Please try using machine.SPI which doesn't require the MASTER parameter (eg `spi = machine.SPI(1)` for SPI1 bus).

zgyarmati, 2017-10-23:
Ah, i see, thanks for the heads-up. In this case it might worth to document this fact (as it's less than obvious for the newcomers that there are 2 SPI APIs), or even provide a ssd1306_test.py script as for lcd160r has one.
I'll send a PR with this test script.

closed 2017-11-03 by zgyarmati