micropython/micropython #3386: [RFC] usocket: way to create sockaddr from binary address

pfalcon, 2017-10-24:
This problem arises when e.g. implementing getaddrinfo() in Python. DNS returns a binary IP address, which somehow needs to be converted to sockaddr, as passed an an argument to most socket functions. There's also an opposite problem - given a sockaddr, split it into components. This is needed e.g. to handle return value of accept (e.g. to print in a human readable form).

Unix usocket actually already tackles the latter problem - it provides sockaddr() function, which takes a sockadr, and return a tuple of (addr_family, bin_addr, port, ...). bin_addr can be converted to textual presentation using inet_ntop().

So, the proposal is to make sockaddr() reversible: passed a single arg, sockaddr, it performs the transformation described above. And if passed more than one arg, it will vice versa construct sockaddr from components.

This function would be mandatory part of usocket API, ports which use tuple-textual sockaddr's, would emulate it.

dpgeorge, 2017-10-26:
> This function would be mandatory part of usocket API, ports which use tuple-textual sockaddr's, would emulate it.

Why the need to add a useless function to bare-metal ports (which are constrained on space)?  Why not just change unix so it already understands addresses in human-readable form, and then remove sockaddr()?

pfalcon, 2017-10-26:
Well, that doesn't seem like a right question. The right (and old) question is: "given that baremetal ports are constrained on CPU performance, and more constrained on RAM, than on ROM, why do they constantly convert txt addresses back and forth to binary, wasting CPU clocks and allocating more memory on that than needed?" The answer is also old: "Some baremetal ports do that for user convenience". It's so old that it's turning into mis-convenience, with some people thinking it's not an exception to technical efficiency, but a normal way to do it.

Socket addresses are real, and there should be full ways to convert different addresses:

sockaddr <=> net_addr_bin <=> net_addr_txt

The right side is covered by inet_pton() and inet_ntop(), the talk is about left side.

pfalcon, 2017-10-26:
Note that it's not too late to introduce full-fledged sockaddr type. It would have str() of "(addr_txt, port)" (i.e. do conversion only if asked by user), apparently would allow to access, maybe even set individual sockaddr components, etc. Actually, unix port started with it like that, but then I thought "why even spend bytes on type structure", so made it just bytes.

Introducing such a type makes sense only if there's an agreement to switch all ports to use it, doesn't make sense to do just for unix port.

dpgeorge, 2017-10-27:
CPython doesn't have socket.sockaddr() so I'd suggest to remove it from uPy and instead make uPy more like CPython in how it handles addresses.

pfalcon, 2017-10-27:
CPython requires megabytes of heap to run and has a reputation of being slow. Nothing of that should apply to MicroPython. Actually, trying to apply it to MicroPython means killing its original idea and flushing a lot of effort put into optimization of it to /dev/null.

> and instead make uPy more like CPython in how it handles addresses.

That's how CPython-compatible "socket" module (which can be/is written in Python) can/does work. The talk here is about "usocket" module, which does things which CPython can't achieve. It needs more optimizations, e.g. I think about accept1() call which returns just a socket instance instead of allocating extra tuple.

And to remind, that's exactly how CPython itself works - it has "_socket" module written in C, and then "socket" written in Python on top of it. To remind, "usocket" was called "_socket" before we came to the general convention of naming modules. As MicroPython also runs on wider variety of systems, it's also helpful to better define interface of "_socket"/"usocket", exactly what we have been doing all this time. In all other aspects, we should follow CPython way, in particular, if there's a pushback  to be "more like CPython", it should be done in socket.py (etc.). Unix port did that all the time, maybe it's time for concerned baremetal ports to do the same.

peterhinch, 2017-10-28:
However you guys resolve this, please bear in mind that a significant use case of the Unix port is to prototype applications ultimately targeted on bare metal. Arguably this is particularly useful for network applications where the bulk of the user code may be hardware-independent.

So it would be good if the ports behaved similarly.

dpgeorge, 2017-10-31:
> in particular, if there's a pushback to be "more like CPython", it should be done in socket.py

The idea with u-modules is that they can be used *without* the need for the corresponding normal-named module that patches things up.  That's the whole idea behind module weak links, so on bare-metal (which can't afford a filesystem and RAM filled with .py wrapper scripts) the normal-named module can link to the u-module and still feel and behave like normal Python.  The unix port should move to using weak links to follow bare-metal, and we shouldn't proliferate the addition of non-CPython functions/methods to u-modules when CPython already has a standard way of doing it which can be made to work.

pfalcon, 2017-11-02:
No, I don't agree that was the exact idea behind "u" modules. Yes, they're usable without corresponding "big" modules, but the whole idea behind them was to provide optimized API where CPython's is too bloated, whenever possible to leverage abstractions CPython provides (like using getaddrinfo() for resolving and just feed the result into connection functions). But when that's not enough, also provide optimized extensions. There're tons of the examples of that already. You perfectly know that I have been spending a lot of effort to optimize various aspects of MicroPython APIs to be able to run efficiently on as small systems as possible, that's in particular why ESP8266 can do something beyond trivialities.

> The unix port should move to using weak links to follow bare-metal

That's as useful as me saying that baremetal ports should drop weak links. Guess why I never thought anything like that? And the talk is not about proliferation, but closing the gaps in the functionality to finalize it all. Then why do I hear responses like that after all the effort which was spent in that direction?

SpotlightKid, 2017-11-29:
> The unix port should move to using weak links to follow bare-metal,

I must say that I agree with @dpgeorge here and pfalcon's counter arguments do not convince me. The whole point of using MicroPython on MCUs for me is that it's *more convenient*. IMHO that sometimes means that defaults behaviours are not implemented in the most optimized manner *if* they make a developer's life much easier.

jonnor, 2024-09-29:
There does seem to be a sockaddr data-type now, and this be the recommended portable way (preferred over textual tuples) - supported by all ports - according to the docs. https://docs.micropython.org/en/latest/library/socket.html#socket-address-format-s

So can we close this issue?

closed 2025-03-03 by jonnor