micropython/micropython #3389: urequests.post failures

dentex, 2017-10-25:
Hello.
I'm trying to receive notifications on my (Android) mobile device from an ESP8266 MCU running Micropython. For this reason I subscribed to a couple of online services exposing some APIs for this task, Pushbullet and Pushed, and I installed the matching apps on my device.

This is what I'm trying:

**Pushbullet:**

```
import json
import urequests

body = "Test Notification"
title = "Pushbullet"
data_sent = {"type": "note", "title": title, "body": body}
API_KEY = 'xxxxxxxxxxxxxxxxxxxxxxxxxxx'

pb_headers = {
    'Authorization': 'Bearer ' + API_KEY, 
    'Content-Type': 'application/json'
}

r = urequests.post(
    'https://api.pushbullet.com/v2/pushes', 
    data=json.dumps(data_sent), 
    headers=pb_headers
)

print(r)
```

Error:

```
ssl_handshake_status: -256
Traceback (most recent call last):
  File "<stdin>", line 11, in <module>
  File "urequests.py", line 104, in post
  File "urequests.py", line 56, in request
OSError: [Errno 5] EIO
```

**Pushed:**

```
import json
import urequests

payload = {
  "app_key": "xxxxxxxxxxxxxxxxxxxxxxxxxxx",
  "app_secret": "xxxxxxxxxxxxxxxxxxxxxxxxxxx",
  "target_type": "app",
  "content": "Remote Mic MCU test from ESP8266"
}

r = urequests.post("https://api.pushed.co/1/push", data=payload)
print(r)
```

Error:

```
Traceback (most recent call last):
  File "<stdin>", line 8, in <module>
  File "urequests.py", line 104, in post
  File "urequests.py", line 74, in request
TypeError: object with buffer protocol required
```

Searching for those errors doesn't get me anywhere useful.

The exact same code snippets work ok on my Linux box (using `requests` instead of `urequests`), but I understand that urequests may have some limitations. Do you have any hint on how to fix this? Thanks.

dpgeorge, 2017-10-26:
For the first issue: the server you are accessing requires server_hostname to be set.  This feature was recently added to urequests so please update to the latest version and try again.  (In f36975b6792f6b702dc9184adcdb6e0d89ce23b4 I added a test to show that this URL does work with uPy.)

For the second issue: the data argument must be a bytes object, urequests doesn't support form encoding.  So either form encode the payload yourself, or use the json arg.

dentex, 2017-10-26:
Hi, thanks for looking into this issue.
Probably I should have mentioned that I was using the latest stable version `esp8266-20170823-v1.9.2.bin`.
In regards to the Pushbullet case, I had a look at https://github.com/micropython/micropython/commit/f36975b6792f6b702dc9184adcdb6e0d89ce23b4 and I modified the headers to be
```
pb_headers = {
    'Authorization': 'Bearer ' + API_KEY,
    'Content-Type': 'application/json',
    'Host': 'api.pushbullet.com'
}
```
Unfortunately I have the same error.
Upgrading to the latest daily build `esp8266-20171026-v1.9.2-195-g328c1e78.bin` instead gives an error importing `urequests`:
`ImportError: no module named 'urequests'`

I still didn't look into the other case (Pushed).

peterhinch, 2017-10-26:
Daily builds don't include some libraries which are part of release builds. You need to install `urequests` from [micropython-lib](https://github.com/micropython/micropython-lib.git).

dentex, 2017-10-26:
I understand. It seems anyway usable only for the Unix port... What about the ESP8266?

dentex, 2017-10-30:
Hey...    :)

dentex, 2017-10-30:
Anyway, I tried the Unix port (I installed `urequests` and `json` with upip) and all works OK.
The problem is now how can I drop the new version on the ESP8266.

dentex, 2017-10-30:
I'm using the latest `urequests.py` from https://github.com/micropython/micropython-lib.git with the firmware version esp8266-ota-20171030-v1.9.2-207-g10b76a96 (that doesn't contain `urequests` by default) and all is working OK.
I mean, I'm importing a file `urequest.py` with the `webrepl_cli` tool, so my `main.py` can call it.
Thanks for the help.

dentex, 2017-11-03:
Hello again,
sorry to re-open the issue. 
This is about the "Pushed" code snippet.
using the `json` keyword as suggested, the request goes, but the response text is
`{"error":{"type":"empty_credentials","message":"You must introduce account credentials to make requests. Please read the documentation available in https:\/\/about.pushed.co\/docs."}}`
...which anyway unjustified since the exact same snippets works on my laptop (using Python3 `request`).

SpotlightKid, 2017-11-29:
I've posted an example on the forum how to send a POST request with `urequests.py` using automatic form encoding of data passed as a dictionary:

https://forum.micropython.org/viewtopic.php?p=22961#p22961

Here's the gist with the code: https://gist.github.com/SpotlightKid/eca9b00239104e8c599b86635f62ab73

I've tested this sucessfully with the unix and the esp8266 port (you need to install the `colections.defaultdict` and the `urequest` module form micropython-lib).

I think this issue can be closed.

If anybody wants to write a HOWTO document for the `urequests` module, I think tha  would be beneficial for many users :)

peterhinch, 2017-11-29:
A tutorial on MicroPython network programming to add to those in "The MicroPython Language" would be great. I'd offer to write it if I had the know-how - I'd certainly be keen to read it :)

loic760, 2018-06-24:
hey, ... how can i install colections.defaultdict to my esp32 ... this isn't like one single.py module..
also `from urllib.parse import urlencode`  returns me no modules named urlencode and it's a blank file in micropython-lib repo :/

SpotlightKid, 2018-06-24:
* Make a folder nymed `collections` on theflash file systemof the esp32 and copy the `defaultdict.py` file into it.
* The `urlencode` module file is in my Gist.

Unknow-per, 2024-04-13:
use this:

r = urequests.post("https://api.pushed.co/1/push", json=payload)
print(r)

it works for me

jonnor, 2024-09-02:
Issue not reproducible and over 5 years without comments. Proposing to close.

closed 2024-09-02 by dpgeorge