micropython/micropython #3394: uselect: Issues with polling stream wrappers

pfalcon, 2017-10-29:
Currently, uselect module accepts only stream which directly implement MP_STREAM_POLL (baremetal ports) or just exact file and socket streams (unix port). However, these streams can be wrapped in (any number of) other streams, e.g. a socket in websocket in SSL, and such wrapped streams don't work with uselect.

That's apparently what being reported in https://github.com/micropython/micropython/issues/3297 .

Solving this on MicroPython level requires a bit of legwork (and associated code), especially given that there're different variants of solution for baremetal vs unix. So, given that uasyncio is the only known component affected, my plan is to try to work it around on its level.

But sooner or later it may affect other users too.

pfalcon, 2017-11-03:
Workaround on https://github.com/micropython/micropython-lib/pull/227 side.

dpgeorge, 2018-07-20:
Support for polling ussl objects on bare-metal (axtls and mbedtls versions) was added in 7a67f057d74e9a16e7ae7bc562e592335145eaee by passing through the ioctl request to the underlying socket object.

Support for polling general objects on unix (including ussl) was implemented in ef554ef9a2708a8f344f22e586f52fb1343ed524

closed 2022-06-23 by dpgeorge