micropython/micropython #3399: Why the module is called _thread instead of uthread?

pfalcon, 2017-10-30:
We don't have the module documented, and it before doing that (even if listing it to acknowledge its existence, until we ourselves forgot what' there), would be nice to rethink subj.

To remind, we settled on the idea that all our builtin modules are targeted for direct use, so they should have "u" prefix instead of "_". That's how _socket was renamed to usocket for example.

dpgeorge, 2017-10-31:
There is no `thread` module in CPython so `uthread` doesn't make sense.  CPython has `threading` and so uPy would have `uthreading`.  `_thread` is like `sys` and `gc`: it's a tightly integrated module that's part of the core.

pfalcon, 2017-11-01:
Ok, sounds good.

closed 2017-11-01 by pfalcon

pfalcon, 2017-11-09:
Docs placeholder added in 579b86451dba202d573c4c22790ebe3d8ddac060.