micropython/micropython #3401: Firmware Updates for KRACK WPA2 Vulnerability

3bb, 2017-10-30:
Espressif has released updates for ESP-IDF, ESP8266 RTOS SDK, & ESP8266 NONOS SDK on their Github page.
Also, the Esp8266 library for Arduino is updated.

https://www.bleepingcomputer.com/news/security/list-of-firmware-and-driver-updates-for-krack-wpa2-vulnerability/

https://www.krackattacks.com

Does 1.9.3v firmware corrects this bug ?

pfalcon, 2017-10-30:
Feel free to submit patches to https://github.com/pfalcon/esp-open-sdk

pfalcon, 2017-11-04:
Updates were posted, to try:

1. The latest esp-open-sdk, build as: `make VENDOR_SDK=2.1.0-18-g61248df`.
2. This patch on uPy side: https://github.com/micropython/micropython/pull/3414

Testing is welcome.

dpgeorge, 2018-05-17:
The v1.9.4 firmware addresses this issue, as long as you used the latest esp-open-sdk.

closed 2018-05-17 by dpgeorge