micropython/micropython #3403: Enable named-groups regex functionality

abencomo, 2017-10-31:
Will it be possible to add support for [named-groups](https://docs.python.org/3/howto/regex.html#non-capturing-and-named-groups) in the ure api? It'll be useful for parsing dynamic URIs/URLs containing named gapturing groups.

pfalcon, 2017-10-31:
Maybe, if you implement support for them (but no guarantee, ure's idea is to be minimal), patches welcome at https://github.com/pfalcon/re1.5/ . As an alternative to re1.5 engine, you can use pcre engine implementation: https://github.com/micropython/micropython-lib/tree/master/re-pcre . There're still no named groups support there, but there, you're definitely welcome to contribute support for them (there're no immediate size constraints there).

dpgeorge, 2022-06-23:
Closing due to inactivity.

closed 2022-06-23 by dpgeorge