micropython/micropython #3405: Add filesize as an optional fourth item of ilistdir() tuple

tomlogic, 2017-10-31:
Context: I'm working on a MicroPython platform with a 1 megabyte SPI flash for storage.  It doesn't make sense to make use of os.stat() on this platform because most of the 10-element tuple holds default values.

I implemented the `os.ilistdir()` method for traversing the directory structure, but I'm surprised by the inclusion of inode (which isn't even populated in the vfs_fat_misc.c implementation I'm looking at) and the lack of a filesize.

Yes, I understand that POSIX `dirent.h` only defines an inode and filename, but MicroPython was designed for embedded environments that typically lack an OS and full POSIX support.

I'd like to propose that before any platforms extend that tuple, we update the definition of `ilistdir()` to define a 4th (optional) entry as filesize so we have consistency with a useful element likely available on most platforms.  For example, the `FILINFO` structure from oofatfs includes `fsize` as its first entry.

As with other fields, an implementation can use `0` if the file's size isn't available (or if the file is a directory).

If we can reach consensus I'm open to working on a PR with the necessary documentation updates.

pfalcon, 2017-10-31:
> Yes, I understand that POSIX dirent.h only defines an inode and filename, but MicroPython was designed for embedded environments that typically lack an OS and full POSIX support.

So, argumentation here seems to go backwards - we work with systems which lack full POSIX support, but we take that as an affinity to lean ("grow") to. While you propose to offer what POSIX doesn't have ;-).

> As with other fields, an implementation can use 0 if the file's size isn't available (or if the file is a directory).

That's not good, because 0 is a valid size.

Otherwise, let's see what @dpgeorge says ;-).

tomlogic, 2017-10-31:
My understanding of POSIX is that it was an attempt to standardize UNIX platforms.  Much of it is based on what was out there, and isn't necessarily an optimal design.  The term "legacy" comes to mind.

If you're looking to run MicroPython on POSIX systems, I guess it's a good standard to lean toward.  But I don't think it's the best choice to guide the design of "a small subset of the Python standard library [...] optimised to run on microcontrollers and in constrained environments."

I like using Python3 for guidance.  But I also like coming up with minimal implementations of useful features if something like implementing `os.scandir()` and `class os.DirEntry` are overkill.

> That's not good, because 0 is a valid size.

My original thought was to use -1 for filesize if unknown, but I was worried it might cause problems when reporting the size of of files larger than 2GB.  ;-)

pfalcon, 2017-10-31:
> But I also like coming up with minimal implementations of useful features if something like implementing os.scandir() and class os.DirEntry are overkill.

That's why ilistdir() was implemented in the first place.

> My original thought was to use -1 for filesize if unknown, but I was worried it might cause problems when reporting the size of of files larger than 2GB. ;-)

There're already problems with reporting sizes more than 1GB on 32-bit system via a small int (1GB - 1 is the max positive value of small int). Indeed, someone might want to reuse negative half of the range to represent sizes from 1GB to 2GB. Then another value should be specified as "n/a".

tomlogic, 2017-10-31:
> That's why ilistdir() was implemented in the first place.

Correct.

> There're already problems with reporting sizes more than 1GB on 32-bit system via a small int (1GB - 1 is the max positive value of small int). Indeed, someone might want to reuse negative half of the range to represent sizes from 1GB to 2GB. Then another value should be specified as "n/a".

I don't think we're limited to a small int, since these are already objects inside of a tuple (and not some lower-level C API).  We could use `None` in the tuple if filesize isn't available, or `SMALL_INT(-1)`.  Devices with large files can still return a `mp_obj_int_t`.

pfalcon, 2017-11-01:
To summarize IMHO, sounds good, as long as its the spec update and there's more than one possible user. Not sure about possible actual code updates in the mainline - there's a queue of more important things...

dpgeorge, 2018-03-07:
I also think it's a good idea to add a 4th element to ilistdir() to return the file size.  In the case that the port doesn't support it then putting -1 in that entry would be a good option, at least compared to `None` a value of -1 is an integer and can be used in expressions the same way as a valid (non-negative) number would.  And `mp_obj_new_int_from_uint()` should be used to support sizes greater than 1GB (this is already what stat does when reporting file sizes).

@tomlogic would you like to work on a PR for this?

tomlogic, 2018-03-07:
@dpgeorge I will work on a PR that includes a documentation update and FAT filesystem (and maybe POSIX?) code changes.  I only have access to a pyboard, and I haven't done much development on it yet.  My focus has been on Digi International's XBee Cellular and Zigbee platforms, but I look forward to getting more code upstream.

dpgeorge, 2018-03-12:
Implemented in 4d3a92c67c9b1550eaf07d06ed74de996ee8fa3b

closed 2018-03-12 by dpgeorge