micropython/micropython #3410: mpy-cross needs a --version option

goatchurchprime, 2017-11-03:
It seems like it's really important when you build the mpy-cross compiler from source that it matches exactly the version of the micropython on the device.  

When you log on to the device you get a detailed version of micropython on the command prompt, like so:
"MicroPython v1.9.3-8-g63826ac5c on 2017-11-01"

Is there a way to get an equally specific version code from the mpy-cross program?

In an attempt to get to the correct one, I am doing:

> git tag --list
> git checkout tags/v1.9.3
> cd mpy-cross
> make

But you can never be sure this is going to be exactly right, and this can be one fiendish source of bugs.

I can see no hidden code supporting this option already in place:
     https://github.com/micropython/micropython/blob/master/mpy-cross/main.c

peterhinch, 2017-11-04:
See #3357.

SpotlightKid, 2017-11-29:
See also this relevant forum discussion: https://forum.micropython.org/viewtopic.php?p=23524#p23524
A `--version` option would indeed be very helpful and should also include the Bytecode version produced by the particular `mpy-cross` binary.

dpgeorge, 2019-05-02:
See #4739 for a patch to do this.

dpgeorge, 2019-05-07:
Option added in 7e90e22ea52665e38138d1e704d5e527439b663c

closed 2019-05-07 by dpgeorge