micropython/micropython #3412: RFC: Provide "development" build/version of MicroPython (unix port)

pfalcon, 2017-11-03:
We've accumulated a number of tickets which don't fit with the main direction/goals of MicroPython (maintaining minimal code size), e.g. https://github.com/micropython/micropython/issues/2926 , https://github.com/micropython/micropython/issues/3009 , etc.

At the same time, if we'd have a special version/build of MicroPython for development testing, that at least would allow to think how to let interested parties prototype these features in the way that would be integratable and testable somewhere.

And we actually have "additional" build in Unix port, micropython_coverage. So, no talk about renaming it so far or anything, just about a path to allow to deal with the issues like above.

pfalcon, 2018-11-17:
The initial PR implementing this is now posted: https://github.com/micropython/micropython/pull/4309 . (Available in my fork otherwise.)

dpgeorge, 2021-04-14:
The `micropython-dev` variant was added some time ago.

closed 2021-04-14 by dpgeorge