micropython/micropython #3421: websocket_helper.py: Make available for all ports including "websocket" module

ccooper21, 2017-11-07:
The `websocket_helper` module is only available as part of the ESP8266 port.  However, this module doesn't include any ESP8266 specific code.  It should be available for any port that includes the `websocket` module.  At a minimum this includes the UNIX port.

Is there a place in the source tree for shared modules written in Python?

pfalcon, 2017-11-07:
Yes, there's a place, but it's in the separate tree - micropython-lib. websocket_helper.py is kept in esp8266 port subdir to make it easier for users to build this webrepl-enabled port. But yes, that doesn't scale, and there's no other choice as to require micropython-lib/ checkout side by side with micropython/.

dpgeorge, 2022-06-23:
`websocket_helper.py` was moved to `extmod/webrepl/` in 7ce1e0b1dc466e48606164aad223c81c93a9cea2