micropython/micropython #3423: Unify interfaces for establishing and using websocket connections

ccooper21, 2017-11-07:
The `websocket_helper` and `websocket` modules are both needed to establish and use a websocket connection.  Each handles a different part of a websocket connection's lifecycle.  The former handles the the initial handshake needed to establish the connection, and the latter handles the on-going use of the connection.  While this technically works acceptably, it creates confusion and makes the use of websockets unnecessarily complex for users.  To effectively use these modules, a user first has to find both modules and determine that they are intended to be used together.  This almost certainly requires an awareness of websocket plumbing.  In lieu of this, the user likely needs to be able to review the code for both modules to gain this knowledge.  This makes the use of websockets unrealistic for less sophisticated users.  Arguably this situation can be improved through documentation alone.  However, unifying the interfaces and making their use more intuitive would be better, and even simplify any subsequent documentation effort.

ccooper21, 2017-11-07:
As of right now, the `websocket_helper` and `websocket` modules do not fully support client connections.  This is being addressed by pull requests #3420 and #3406.  For reference, all see issue #3421.

pfalcon, 2017-11-12:
The latest outlook at this is at https://github.com/micropython/micropython/pull/3413#issuecomment-341898639 , point 1. Summary: Python is the language of convenience *wrappers*. Either single party or anyone can make a module which exposes API the see best, then anyone not agreeing can make their own instead. Then, with MicroPython's goals in mind, MicroPython should use completely different criteria for API design, that's: allow to do as much as possible with as little as possible resources (following an insight that there're scarce resources, and they belong to their owners, not to MicroPython). Then interested parties are welcome to provide convenience wrappers on top of that (while letting people who're really short of resource to use the baseline resource-saving version).

That's why https://github.com/micropython/micropython/pull/3420#issuecomment-343741241 suggests to add more modules, not less, so users can mix and much what they want and need.

pfalcon, 2017-11-12:
Also see https://github.com/micropython/micropython-lib/pull/228. The author of that patch apparently thinks that the right way to use websockets are with uasyncio. Do you agree? No? Then you'll be writing your own wrapper. The point is that underlying, internal implementation should support different paradigms, while being the most minimal.

pfalcon, 2017-11-12:
Finally, websocket functionality is not documented exactly because it's API is not settled, starting from "websocket" module. That can easily become `_websocket`. Maybe even `_ws` is someone will argue for it too well.

dpgeorge, 2022-06-23:
Closing due to inactivity.

closed 2022-06-23 by dpgeorge