micropython/micropython #3424: Reaching VFS object without mounting it in vfs_fat_ramdisk.py

husigeza, 2017-11-07:
Hello,

When running the tests in vfs_fat_ramdisk.py I noticed the following behavior:
At line 91 the file system is being unmounted, which means the synchronization object of the File System is removed/deleted. 
At line 93 a new Virtual File System object is created "from" the block device. 
At line 94 the directories of the previously created VFS object is read (the FS behind the VFS object has not been mounted before). The "ilistdir()" function ends up trying to take the synchronization object (semaphore) which does not exist because it has not been created via the "mount()" process. The port of the cc3200 device in this case would crash, because the relevant FreeRTOS function asserts on the existence of the synchronization object. I have not tired the cc3200 port yet, but my port is very similar to it and it crashes. 
Do I miss something here ? Probably some configuration options are incorrect ?

Thanks!

dpgeorge, 2017-11-16:
Thanks for the report.  Indeed there is a problem that is only evident when the `_FS_REENTRANT` option is enabled.  It should be fixed by #3443 (see comments there on how it fixes it).  Please can you test that code to see if it works in your case?

husigeza, 2017-11-16:
Hello,

I tried your solution and it fixed the problem. However I noticed the following:
Let's assume we have an SD card with a FAT filesystem:
1. Mount the block device under /sd with os.mount()
2. Create a new VFS object with: fs = os.VfsFat(sd)
3. Add a new directory with: os.mkdir("/sd/test1")
4. Add a new directory through the VFS object: fs.mkdir("test2") // Originally the system would crash here
5. List the directories under /sd with os.listdir(): it will only show test1
6. List the directories of the VFS object with fs.ilistdir(): it will only show test2
7. Do an os.umount() (unecessary to do fs.umount() as it does nothing)
8. List the directories under /sd: it will list both test1 and test2

So it seems an inconsistent state happens when reading the same files system/device, depending on the object used for reading.

dpgeorge, 2017-11-17:
> So it seems an inconsistent state happens when reading the same files system/device, depending on the object used for reading.

The issue is that you are accessing the same block device from 2 different VFS objects.  The VFS object has some state (cache) associated with it so changes made on the filesystem via one of these objects may not be seen by the other object until the cache/state is flushed and reloaded (not something you can force to happen).

This is not really how the VFS was intended to be used.  If you want to have a VFS object and the same thing mounted at a specific point then do the following:
```python
fs = os.VfsFat(sd)
os.mount(fs, '/sd')
# now access via os will go through fs object, so everthing is in sync
```

husigeza, 2017-11-19:
Alright, then the bug is fixed.

closed 2017-11-19 by husigeza

dpgeorge, 2017-11-20:
Ok, great.  PR #3443 is now merged.

PR #3443 (2017-11-20): extmod/vfs_fat: Mount FatFS on creation so VFS methods can be used.
It's possible to use the methods (eg ilistdir) of a VFS FatFS object
without it being mounted in the VFS itself.  This previously worked but
only because FatFS was "mounting" the filesystem automatically when any
function (eg f_opendir) was called.  But it didn't work for ports that used
synchronisation objects (_FS_REENTRANT) because they are only initialised
via a call to f_mount.  So, call f_mount explicitly when creating a new
FatFS object so that everything is set up correctly.  Then also provide a
finaliser to do the f_u [... cut]