micropython/micropython #3426: uasyncio.await reader.read()(in progress)  remains indefinitely pending

DomCharn, 2017-11-08:
Hello,
after calling await asyncio.open_connection and getting  writer,reader ( Streamreader & Streamwriter)
a close socket occured, socket status became -16.
In this situation a 'in progress' await reader.read remains indefinitely pending
Is there a mean to let uselect.poll detect closed(-16?)  socket on select.POLLIN  poll.registered socket ?
In situation, 
    reader.s gives <socket state=-16 ...
    writer.s.send('ss') returns EBADF
    reader.s. recv (..) returns EBADF    
Thanks for your return
Regards

pfalcon, 2017-11-08:
Is this a bug report? If so, bug report should include: 1) description of where the problem happens (mirrored in the title if needed); 2) exact steps to reproduce.

More info: https://www.chiark.greenend.org.uk/~sgtatham/bugs.html

dpgeorge, 2020-06-09:
This bug should not be in the latest uasyncio, so I'll close the issue.  Feel free to reopen if the bug persists, with detail how to reproduce it.

closed 2020-06-09 by dpgeorge