micropython/micropython #3427: SmartConfig ESP8266 and ESP32

zaforic, 2017-11-08:
Add smartconfig capability to micropython for esp8266 and esp32

Add the following to modesp.c
{ MP_ROM_QSTR(MP_QSTR_smartconfig), MP_ROM_PTR(&mp_module_esp_smartconfig) },

Create smartconfig python wrapper. An old example is at https://github.com/mianos/micropython/blob/master/esp8266/mod_esp_smartconfig.c

xiaoATT, 2019-10-10:
how can i use smartconfig now

D1ver, 2023-01-06:
No update?

hzbd, 2024-10-31:
There's a third-party open source implementation. Unfortunately, its documentation is in Chinese. A pull request has been submitted, but progress has been slow.

https://gitee.com/walkline/micropython-smartconfig-cmodule

https://github.com/micropython/micropython/pull/13658

Walkline80, 2024-10-31:
> Unfortunately, its documentation is in Chinese.

Thanks for the mention, the Chinese docs just describe how to compile SmartConfig as a user c module and how to use it (ESP Touch and WeChat applets)

> A pull request has been submitted, but progress has been slow.

Actually I don't know how to push the PR process so I can only wait for now....

Walkline80, 2024-10-31:
For the record, that PR currently implements the esp32 port only, not esp8266.