micropython/micropython #3428: micropython.schedule fails when passed a bound method

peterhinch, 2017-11-09:
Test conditions
```
MicroPython v1.9.2-153-g7c38ae0 on 2017-10-24; PYBv1.1 with STM32F405RG
```
The following fails
```python
from micropython import schedule, alloc_emergency_exception_buf
alloc_emergency_exception_buf(100)

class Foo():
    def __init__(self):
        tim = pyb.Timer(4)              # create a timer object using timer 4
        tim.init(freq=2)                # trigger at 2Hz
        tim.callback(self.cb)

    def cb(self, t):
        schedule(self.bar, 0)

    def bar(self, _):
        print('Bound method')

foo = Foo()
```
Outcome:
```
>>> uncaught exception in Timer(4) interrupt handler
Traceback (most recent call last):
  File "<stdin>", line 11, in cb
MemoryError: memory allocation failed, heap is locked

>>> 
```

robert-hh, 2017-11-10:
Look at Dave#s comment here: https://forum.micropython.org/viewtopic.php?f=2&t=4027&p=23118#p23118

peterhinch, 2017-11-10:
Thanks @robert-hh I'd just spotted that. The following version works
```python
from micropython import schedule, alloc_emergency_exception_buf
alloc_emergency_exception_buf(100)

class Foo():
    def __init__(self):
        def bm(_):
           print('Bound method')
        self.bar = bm
        tim = pyb.Timer(4)              # create a timer object using timer 4
        tim.init(freq=2)                # trigger at 2Hz
        tim.callback(self.cb)

    def cb(self, t):
        schedule(self.bar, 0)

foo = Foo()
```
This strikes me as counterintuitive. If this is a permanent behaviour of MicroPython I'll submit a PR amending the "writing interrupt handlers" doc to suit.
@dpgeorge @pfalcon What is the official view on this behaviour?

dhylands, 2017-11-10:
Note that self.cb (that you're passing to tim.callback) is also a bound method. Any occurrence of self.function_name is a bound method. Foo.cb would be the unbound function

pfalcon, 2017-11-10:
@peterhinch : I don't all the above should come as surprise to an experienced MicroPython user. Bound method is a pair of (object, method), as it's not a single value, it needs to be allocated, so when allocation isn't possible, you can't take a ref of bound method (but can call it, that's special optimization in the compiler, which @dpgeorge explained yet in the original KS updates docs. So, it has nothing to do with .schedule() and has everything to do with interrupts.

But yes, for novices, the docs need to mention this case with schedule() specifically. When it comes to document schedule(), a note on that might be included in the reference docs too.

peterhinch, 2017-11-11:
OK, thanks both. I'll submit PR's against the docs.

SpotlightKid, 2017-11-29:
Fix is merged. This issue can be closed, right?

peterhinch, 2017-11-29:
Yes.

closed 2017-11-29 by peterhinch