micropython/micropython #3432: RFC Re-standardize to SHA1 as the "default" hashing algorithm

pfalcon, 2017-11-12:
We initially selected SHA256 as the "current", "latest generation" hashing algorithm to make available in MicroPython "by default", but seems it's actually "too fresh", and there're useful technologies which use earlier hashes, e.g. Websockets use SHA1 for handshake.

So, the proposal is to pull in a SHA1 implementation, which can be built regardless of external dependencies. (SHA256 can just stay of course, not need to remove it.)

ccooper21, 2017-11-12:
Is this past decision the reason that the `uhashlib.sha256` class is implemented by standalone code as opposed to the `uhashlib.sha1` class that depends on the implementation provided by the axTLS or mbed TLS library?

I'm assuming the main consideration here is the code footprint.  If my assumption is correct, and the axTLS or mbed TLS libraries are in use, shouldn't the SHA256 implementation just use the axTLS or mbed TLS implementation like the SHA1 implementation?  I'm wondering if the SHA1/SHA256 implementations get much (any?) use in environments that don't have networking support, and hence don't have TLS support.  The only user of the `uhashlib` module in the `micropython` code base is the `websocket_helper` module.  Similarly, in the `micropython-lib` code base the only user is the `hmac` module, which actually uses the `hashlib` module provided by `micropython-lib`, but should be using the `uhashlib` module.  Hence, maybe there is no practical need include a standalone implementation.

In other words, maybe the `uhashlib` module should only be enabled when a TLS/SSL implementation is available, and it should include both SHA1 and SHA256 implementations.

pfalcon, 2018-08-19:
> In other words, maybe the uhashlib module should only be enabled when a TLS/SSL implementation is available, and it should include both SHA1 and SHA256 implementations.

This makes no sense. Hashing is a general-purpose data processing, not tied into/with TLS (excep implementation-wise).

pfalcon, 2018-08-19:
Also, dismissing MD5 was a bit naive - it's still used as parts of various protocols, e.g. HTTP Digest auth. So if interoperability is important, it should possible to make it available too.

pfalcon, 2018-08-19:
uhashlib.md5() implementation posted as https://github.com/micropython/micropython/pull/4063

rolandvs, 2018-08-20:
Isn't SHA-1 a little different from MD5?
I have added SHA-1 much in the same way as SHA-256 in uhashlib and added it. It is independent of axTLS, TLS or mbed stuff, etc.
Maybe it can be used to provide SHA1 in a way Paul (@pfalcon) suggested.
I'm happy to put it in a PR if that is considered useful.

pfalcon, 2018-08-20:
> I have added SHA-1 much in the same way as SHA-256 in uhashlib and added it. It is independent of axTLS, TLS or mbed stuff, etc.

What's needed is not just add a random SHA-1 implementation found out there, but survey a bunch of implementations, and choose the *smallest* by the code size (also with compatible license, etc., etc.). That what was done for SHA-256 when its support was initially implemented. The project it was taken from would be definitely a good candidate to also look for SHA-1, but it was the best candidate found 3-4 years ago. The scene should be re-surveyed, maybe even smaller implementation surfaced.

dhylands, 2018-08-20:
One candidate to consider is https://github.com/rweather/arduinolibs/ which includes a bunch of Crypto libraries designed for the Arduino (and licensed under MIT). SHA1 can be found in the libraries/CryptoLegacy/src

While these are C++ files, they would be fairly easy to convert to C.

dpgeorge, 2022-06-24:
Closing due to inactivity.

closed 2022-06-24 by dpgeorge