micropython/micropython #3437: ESP8266: Unable to read or write on files

yeyeto2788, 2017-11-14:
Hello there,

I'm facing a problem when trying to read or write a file on the ESP8266 the MCU have the 1.9.3 firmware on it.

```
MicroPython v1.9.3-8-g63826ac5c on 2017-11-01; ESP module with ESP8266
Type "help()" for more information.
>>> import os
>>> f = open('messages.txt','r')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
>>> f = open('./messages.txt','r')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
>>> f = open('./messages.txt','w')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 13] EACCES
```

As you can see I get 2 different errors one is the `[Errno 2]` when I try to read the file and the `[Errno 13]` when I try to write data on the file.

Here is what I have on the MCU and I can only access to the `boot.py`
```
>>> import os
>>> os.listdir()
['boot.py', 'style.css', 'messages.txt']
>>> with open('boot.py') as f:
...     a = f.read()
...     print(a)
...
...
...
# This file is executed on every boot (including wake-boot from deepsleep)
#import esp
#esp.osdebug(None)
import gc
#import webrepl
#webrepl.start()
gc.collect()
```

robert-hh, 2017-11-14:
Errno 2 is "file not found". That looks like a corrupt file system. Did you try to erase the flash and reload the firmware & files?
Side note: Questions like these are better asked in the forum. (https://forum.micropython.org/)

yeyeto2788, 2017-11-14:
I did it before creating the issue here, 

Here you have the output:

```
win32@RTCMW0941M MINGW64 ~/Downloads
$ esptool.py --port COM10 --baud 460800 erase_flash
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.2s
Hard resetting...

win32@RTCMW0941M MINGW64 ~/Downloads
$ esptool.py --port COM10 --baud 460800 write_flash --flash_size=detect 0 esp8266-20171101-v1.9.3.bin
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0040
Compressed 600888 bytes to 392073...
Wrote 600888 bytes (392073 compressed) at 0x00000000 in 8.9 seconds (effective 540.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
```

And also doing a firmware check says that everything is ok:
```
>>> import esp
>>> esp.check_fw()
size: 600872
md5: 35e285a80516e70242ebf7d780d6c70f
True
```

That's the main reason why I opened the issue here, but thank you very much for the quick reply.

Best Regards.

peterhinch, 2017-11-14:
You will get a better response raising issues like this on [the forum](https://forum.micropython.org/) or simply doing a forum search. The usual solution to filesystem problems is to erase flash before installing the firmware.

robert-hh, 2017-11-14:
Does the detected flash size match the one on the board? Otherwise, I see no problem. I just checked. The flash command the makefile issues is:
```
esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash --verify --flash_size=4MB --flash_mode=qio 0 build/firmware-combined.bin
```
Obviously, I'm using Linux, and a self-built image. The flash mode may be dio, depending on your device,

peterhinch, 2017-11-15:
Other possibilities are power supply issues or a poor quality flash device.

yeyeto2788, 2017-11-15:
@robert-hh the flash size it is actually 4MB as the `esptool` detects it, I did erase flash before flashing the firmware. I'm running the firmware on a Wemos D1 Mini which is one of the boards that many people use and I also try to do the same on a Witty (Chinese ESP8266 board) and I got the same results.

@peterhinch I would check the power supply and voltage regulator but if that was the case I shouldn't be even able to read serial or the `boot.py` file in the board, right? 

Thank you guys for help out here.

robert-hh, 2017-11-15:
I'm using the Wemos D1 myseld, and I think @peterhinch too, without any problems on reading or writing files. Then only thing which I not did is using the same firmware build like you. I can try that this night, using this build: http://micropython.org/resources/firmware/esp8266-20171114-v1.9.3-20-g964bf935.bin

yeyeto2788, 2017-11-15:
@robert-hh  I'll try that build.

The weird part is that using the Adafruit's ampy tool `version 1.0.2` I'm able to load a file into the board and read the same file within the Micropython REPL itself but trying to read a file created **from the REPL** it does show the error.

ampy commad `ampy -p COM10 put messages.txt`

```
MicroPython v1.9.3-8-g63826ac5c on 2017-11-01; ESP module with ESP8266
Type "help()" for more information.
>>> import os
>>> os.listdir()
['boot.py', 'messages.txt']
>>> os.mkdir('test.py')
>>> os.listdir()
['boot.py', 'messages.txt', 'test.py']
>>> f = open('test.py','r')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
```

robert-hh, 2017-11-15:
So you are creating a directory with the name "test.py" and try to read it. That's the problem. You cannot read a directory like a  file.

yeyeto2788, 2017-11-15:
Ohhh S/*-%$... 

That's the problem, I'm so embarrassed for the mistake. 

Here you have the output.
```
>>> import os
>>> f = open('testing.txt','w')
>>> f.write("hello")
5
>>> f.close()
>>> os.listdir()
['boot.py', 'messages.txt', 'test.py', 'testing.txt']
>>> f = open('testing.txt','r')
>>> print(f.read())
hello
>>> f.close()
```

I'll close the issue then.

Thank you very much for the help @robert-hh , @peterhinch and once again sorry for the misunderstanding

closed 2017-11-15 by yeyeto2788