micropython/micropython #3439: On some esp8266 hardware, utime.time is not using seconds?

larsks, 2017-11-14:
I am running MicroPython (`MicroPython v1.9.3-8-g63826ac5c`) on two different esp8266 boards:

- A [DigiStump Oak](https://digistump.com/wiki/oak)
- A [Wimo Mini D1](https://wiki.wemos.cc/products:d1:d1_mini) (possibly a clone...)

On all the Wimo devices, if I run:

```
>>> import time
>>> a = time.time()

```
Then wait about 10 wall clock seconds and run:

`>>> b = time.time()
`

Then the difference is about what you would expect:

`>>> b-a
`

But on the Digistump Oak, the same code running the same version of MicroPython yields a much different result:

```
>>> b-a
11738

```

The two types of boards report the same value for `machine.freq()`:

```
>>> machine.freq()
80000000

```

I've rewritten my code to use `time.ticks_ms` instead, but what could be causing the different behavior of `time.time` on these boards?

dpgeorge, 2017-11-15:
The `time.time()` function uses the RTC, which is calibrated at runtime.  The `time.ticks_ms()` function uses an internal microsecond timer based on the CPU's crystal.  So they are different sources.  It may be that the calibration is way out on the Oak board.  Is your Oak board using a standard ESP-12 module, or something custom?

larsks, 2017-11-15:
It looks like the answer is "something custom" (the chip is an "Acorn" module, http://digistump.com/products/149).

dpgeorge, 2017-11-15:
> the chip is an "Acorn" module

In that case you'll need to find the schematics for the Acorn and check them against the ESP-12 to see why it differs.  It may be using a different crystal.

larsks, 2017-11-15:
Yeah, fair enough. Thanks!

closed 2017-11-15 by larsks