micropython/micropython #3440: Exposing number of matches in a match object?

larsks, 2017-11-15:
The `Match` objects returned by `re.match` and `re.search` internally stores the number of match groups:

>>> r = re.compile('hello my ([^ ]+) is ([^ .]+).')
>>> m = r.search('hello my name is lars')
>>> m
<match num=3>

This value isn't currently exposed in Python.  The workaround for iterating through available matches, which we see in some of the tests, is sort of clunky:

```
def print_groups(match):
    print('----')
    try:
        i = 0
        while True:
            print(match.group(i))
            i += 1
    except IndexError:
        pass

```
I've read through the [contributor guidelines](https://github.com/micropython/micropython/wiki/ContributorGuidelines), and since there exists a workaround I understand that simply exposing the match count may not be a priority.  Would you accept a documentation pr that includes some variant of the above example in the `ure` module documentation?

robert-hh, 2017-11-15:
Most likely you're right, even if the C implementation would be rather simple, like:
```
STATIC mp_obj_t match_count(mp_obj_t self_in) {
    mp_obj_match_t *self = MP_OBJ_TO_PTR(self_in);
    return MP_OBJ_NEW_SMALL_INT(self->num_matches);
}
MP_DEFINE_CONST_FUN_OBJ_1(match_count_obj, match_count);
```
plus the definition of the method, which is another line in the code.

dpgeorge, 2022-06-23:
An re match object now supports the `groups()` method which returns a tuple of all groups.  That can help in this situation.

AFAIK, CPython does not support getting the number of groups in a match (without doing `len(match.groups())`.  So we wouldn't want to add it in MicroPython.

> Would you accept a documentation pr that includes some variant of the above example in the `ure` module documentation?

Yes that would be acceptable.