micropython/micropython #3441: Custom exceptions don't support keyword arguments

larsks, 2017-11-15:
I have code that looks like:

```
class MyException(Exception):
    def __init__(self, details=None):
        self.details = details

def error():
    raise MyException(details='This is a test')

```

Calling the `error` function results in:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "extest.py", line 8, in error
TypeError: function does not take keyword arguments

```

I think this is mostly a documentation bug.  The [builtin types][] documentation has two similar but not quite identical entries:

- [User-defined attributes for builtin exceptions are not supported][1]
- [Exception.__init__ raises TypeError if overridden and called by subclass][2]

I'm unclear if this is actually just a new expression of one of the existing cases, or if there should be a new entry along the lines of "Keyword arguments on user-defined exceptions are not supported".

[1]: https://docs.micropython.org/en/latest/pyboard/genrst/builtin_types.html#user-defined-attributes-for-builtin-exceptions-are-not-supported
[2]: https://docs.micropython.org/en/latest/pyboard/genrst/builtin_types.html#exception-init-raises-typeerror-if-overridden-and-called-by-subclass
[builtin types]: https://docs.micropython.org/en/latest/pyboard/genrst/builtin_types.html

dpgeorge, 2017-11-16:
This is actually related to #3323 and is a limitation of uPy at the moment: that if you subclass a native class (in this case `Exception`) then the signature of the subclassed constructor must match the signature of the native base.

dpgeorge, 2017-12-12:
This is now partially fixed by d32d22dfd79439e07739166eaa3f7e41466c7ec8.  At least now there is a work-around that is compatible with CPython: explicitly call the parent (super) constructor like this:
```python
class MyException(Exception):
    def __init__(self, summary, details=None):
        self.summary = summary
        self.details = details
        super().__init__(summary)
```
That will work as expected.

jonnor, 2024-09-29:
A reasonable workaround seems to be available. Is it still relevant to keep this issue open?

closed 2025-06-10 by dpgeorge