micropython/micropython #3442: When is a string not a string? Problems with the config method on network interfaces.

larsks, 2017-11-15:
I'm running MicroPython v1.9.3-8-g63826ac5c on an esp8266.

Given:

`>>> ap_if = network.WLAN(network.AP_IF)
`

This works:

```
>>> ap_if.config('essid')
'MicroPython-35a0d8'

```

And this works:

```
>>> key = 'essid'
>>> ap_if.config(key)
'MicroPython-35a0d8'

```

But this fails:

```
>>> key1 = 'essid'
>>> ap_if.config(key1)
'MicroPython-35a0d8'
>>> somevar = b'essid'
>>> key2 = str(somevar, 'ascii')
>>> ap_if.config(key2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: unknown config param
```

Even though:

```
>>> type(key1) == type(key2)
True
>>> key1 == key2
True
>>> repr(key1) == repr(key2)
True

```

What's going on here?

larsks, 2017-11-15:
And for completeness, wrapping that final example in another call to `str()` works:

```
>>> somevar = b'essid'
>>> key2 = str(somevar, 'ascii')
>>> key3 = str(key2)
>>> ap_if.config(key3)
'MicroPython-35a0d8'

```

dpgeorge, 2017-11-16:
Thanks for the report.  This is a consequence of network.config() checking its incoming parameters against interned string identifiers, not actual string data.  Looking into this actually opened a small can of worms, because it turns out that there are lots of places where one can create a str object that is a duplicate of an existing interned object (eg `' essid '.split()[0]` will also fail in the above example).

In the interest of reducing heap allocation wherever possible, I've now fixed all the cases where new str objects could be returned without first checking if they duplicate an existing interned str.  Such a fix has also fixed this issue; see 1f1d5194d775ad996f1d341c1a44b56af7ea4d4c and 8d956c26d150749375115346f4ca319455107587

closed 2017-11-16 by dpgeorge

larsks, 2017-11-16:
Thanks! That fixed it.