micropython/micropython #3445: freezing an .mpy with mpy-cross -s suffix = *.mpy results in frozen_mpy.c having a DOT prepended to "__lt_module_gt_"

adritium, 2017-11-17:
1. Create empty `main.py`
2. "../../mpy-cross/mpy-cross.exe" -o main.mpy -s main.**mpy** main.py
Note the `-s` param ends in **.mpy**, not the normal **.py**
3. python "../../tools/mpy-tool.py" -f -q build\genhdr\qstrdefs.preprocessed.h main.mpy > frozen_mpy.c
will result in a dot placed where it shouldn't be i.e. search for `.__lt_module_gt_` (it's in 4 places)
```

#include "py/mpconfig.h"
#include "py/objint.h"
#include "py/objstr.h"
#include "py/emitglue.h"

#if MICROPY_OPT_CACHE_MAP_LOOKUP_IN_BYTECODE != 0
#error "incompatible MICROPY_OPT_CACHE_MAP_LOOKUP_IN_BYTECODE"
#endif

#if MICROPY_LONGINT_IMPL != 2
#error "incompatible MICROPY_LONGINT_IMPL"
#endif

#if MPZ_DIG_SIZE != 16
#error "incompatible MPZ_DIG_SIZE"
#endif

#if MICROPY_PY_BUILTINS_FLOAT
typedef struct _mp_obj_float_t {
    mp_obj_base_t base;
    mp_float_t value;
} mp_obj_float_t;
#endif

#if MICROPY_PY_BUILTINS_COMPLEX
typedef struct _mp_obj_complex_t {
    mp_obj_base_t base;
    mp_float_t real;
    mp_float_t imag;
} mp_obj_complex_t;
#endif

enum {
    MP_QSTR_main_dot_mpy = MP_QSTRnumber_of,
};

extern const qstr_pool_t mp_qstr_const_pool;
const qstr_pool_t mp_qstr_frozen_const_pool = {
    (qstr_pool_t*)&mp_qstr_const_pool, // previous pool
    MP_QSTRnumber_of, // previous pool size
    1, // allocated entries
    1, // used entries
    {
        (const byte*)"\xc4\x08" "main.mpy",
    },
};

// frozen bytecode for file main.mpy, scope main._<module>
STATIC const byte bytecode_data_main.__lt_module_gt_[16] = {
    0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07,
    MP_QSTR__lt_module_gt_ & 0xff, MP_QSTR__lt_module_gt_ >> 8,
    MP_QSTR_main_dot_mpy & 0xff, MP_QSTR_main_dot_mpy >> 8,
    0x00, 0x00, 0xff,
    0x11, 
    0x5b, 
};
const mp_raw_code_t raw_code_main.__lt_module_gt_ = {
    .kind = MP_CODE_BYTECODE,
    .scope_flags = 0x00,
    .n_pos_args = 0,
    .data.u_byte = {
        .bytecode = bytecode_data_main.__lt_module_gt_,
        .const_table = NULL,
        #if MICROPY_PERSISTENT_CODE_SAVE
        .bc_len = 16,
        .n_obj = 0,
        .n_raw_code = 0,
        #endif
    },
};

const char mp_frozen_mpy_names[] = {
"main.mpy\0"
"\0"};
const mp_raw_code_t *const mp_frozen_mpy_content[] = {
    &raw_code_main.__lt_module_gt_,
};

```

dpgeorge, 2017-11-20:
> I understand you'd want to have -s with .py usually and I wasn't trying to break mpy-cross. If for some reason mpy-cross -s *.mpy isn't allowed

How did you run into this error, was it just a typo in your build script, or did you intend to have the file extension ".mpy"?  Probably a file with a ".mpy" extension won't import because the import machinery requires ".py" for frozen scripts.

adritium, 2017-11-20:
@dpgeorge

Yes it was a typo. The problem is that there’s no feedback that `-s *.mpy` is invalid: `mpy-cross` produces an `.mpy` with no error messages and `mpy-tool` produces frozen mpy code also with no error message. 

Sounds like you’re saying that `mpy-cross` should just enforce a `.py` suffix for `-s` parameter.

dpgeorge, 2017-11-20:
> Sounds like you’re saying that mpy-cross should just enforce a .py suffix for -s parameter.

That's the intended use at the moment.  But it may be in the future that other extensions are allowed, so the fix/warning would find a better home in mpy-tool.py.

adritium, 2017-11-20:
Is it the intended behavior to produce that DOT in the C-code when `-s` = *.mpy?

dpgeorge, 2017-11-20:
> Is it the intended behavior to produce that DOT in the C-code when -s = *.mpy?

No.  It's a consequence of a this line: https://github.com/micropython/micropython/blob/master/tools/mpy-tool.py#L533 , which is a very quick-and-dirty way to convert the (.py) filename to a valid C identifier.  It's easy to fix that line to be more general and safer.

jonnor, 2024-09-29:
Fixed welcomed for this minor issue. If no contributions come in, we might close the issue, as the problem is very minor.