micropython/micropython #3451: uselect polling: poor performance on ESP8266

jacob019, 2017-11-21:
uselect is performing poorly on the ESP8266.  The poll method hangs for a while even when the tcp receive buffer is full.  uasyncio takes a big performance hit, with an 80kb file taking about 15 seconds to serve with picoweb, while <2s serve times are possible.

When the poll method is in a tight loop of native code it performs well.  It also performs well in simple tests without any function calls in the loop.  Add any complexity and the performance tanks.

I have attached a test case that shows the difference in time spent waiting on the poll function between identical native and non-native code.  Other differences in execution speed between native and non-native code should not be reflected.  It serves a little over 100kb of spaces when the connection is received.
[uselect-test.py.zip](https://github.com/micropython/micropython/files/1493057/uselect-test.py.zip)

Here's a shell command to make the requests:
for ((n=0;n<10;n++)); do sh -c 'nc 10.0.0.172 9999 > /dev/zero'; done

The timed output is printed on the REPL:
Not Native: 12669ms
Native: 3646ms
Not Native: 15057ms
Native: 3725ms
Not Native: 11584ms
Native: 3922ms
Not Native: 11080ms
Native: 4232ms
Not Native: 11839ms
Native: 3939ms

Here we see a 2X-3X difference, but for more complex work loads the difference is a dramatic, 5X-7X.

dpgeorge, 2017-11-21:
Thanks for the report and reproducible test.  Running the script, I didn't see as big a difference as above but instead got:
```
Not Native: 10163ms
Native: 5661ms
Not Native: 7100ms
Native: 6577ms
Not Native: 6857ms
Native: 6545ms
Not Native: 12902ms
Native: 5462ms
Not Native: 8714ms
Native: 5846ms
```
So, the problem here looked very similar to the TCP upload issue discussed in https://github.com/micropython/micropython/pull/3379#issuecomment-340368131 .  And indeed it had the same fix.  Please see #3452 where I took a commit from #3379 and put it in its own PR.  With this small patch at #3452 applied I now get the following output running the above test:
```
Not Native: 617ms
Native: 515ms
Not Native: 512ms
Native: 894ms
Not Native: 549ms
Native: 546ms
Not Native: 566ms
Native: 506ms
Not Native: 579ms
Native: 469ms
```

jacob019, 2017-11-22:
10X improvement, that's a game changer. Thank you for the amazing work. I wish there was a way to make a small cash donation to the project.

closed 2017-11-22 by jacob019

dpgeorge, 2017-11-22:
Ok, great that it works!

> I wish there was a way to make a small cash donation to the project.

There's actually an issue for that: #2515, but unfortunately there was no progress there...