micropython/micropython #3453: Software drivers API

aykevl, 2017-11-22:
Some sensors and actuators need drivers written in C. Examples are: WS2812 (NeoPixel), APA102, DHT family. The protocols for these devices have tight timing requirements or have other requirements which make it difficult or impossible to implement them in Python.

On the ESP8266 these are implemented in the `esp` module and are not documented. Instead, most programming should be done using the relevant high-level Python wrapper classes.

Unfortunately, this makes these wrapper classes device dependent. It would be nice to make them universal. So here is a proposal for a drivers module. It can be named e.g. `drivers` or `_drivers` to indicate it's a low level API. The functions inside should be as minimal as possible, only exposing the bare minimum required in C.

```python
_drivers.ws2812_write(pin, buffer)
_drivers.apa102_write(pin, buffer)
_drivers.dht_read(pin, buffer)
```

The pin in this case is a pin as defined in the machine module and the buffer can be anything following the buffer protocol (?), for example a `bytearray` or `array.array`. Note that I have used `ws2812_write` instead of `neopixel_write` as the latter is a brand name (from Adafruit) and the protocol is not constrained to these LED strips. Otherwise it is mostly identical to the functions found in the `esp` module.

Another consideration: what about bitbanged implementations of common peripherals like UART? These might be useful and possibly more suited for this module than the machine module. In case there is a desire to implement such protocols in software.

The motivation for this API is that I'm contributing to the nrf port of MicroPython and am wondering where to put these drivers.

dpgeorge, 2017-11-22:
> Examples are: WS2812 (NeoPixel), APA102, DHT family, Dallas 1-wire protocol. There are probably many others. The protocols for these devices have tight timing requirements or have other requirements which make it difficult or impossible to implement them in Python.

I have in the past tried to invent a common interface/function for all these protocols but didn't succeed.  Inspired by `machine.time_pulse_us()` the idea would be to make a generic "bit bang" function like:
```python
machine.out_pulse_train_us(pin, bytes, time_lo, time_hi)
```
that would convert the `bytes` argument to a bit stream on the given pin with the given timing.  But the timing requirements for WS2812 are so tight that I doubt it would work (at least not on esp8266).  One could make the function more specialised, eg by saying that a "0" is converted to a pulse with a certain off/on time, and a "1" has a different off/on time, like:
```python
machine.out_coded_pulse_train(pin, bytes, time_0_lo, time_0_hi, time_1_lo, time_1_hi)
```
That might work for WS2812.  But then you also need to read pins for some protocols (like 1-wire) and it gets really messy.

Note that APA102 can probably be driven using plain old SPI, even software SPI (maybe with an extension that the MISO pin is unused).

> Unfortunately, this makes these wrapper classes device dependent. It would be nice to make them universal.

Yes definitely.  Given that my idea above to make a generic bit-bang function is probably not going to work, the next best thing would be to unify all these individual drivers, make them generic and available on all ports that want/need them.

I'm not sure that `drivers` is the right name becase it's pretty high-level and generic and should be reserved for something more user-facing.  Maybe something like `lldrivers` for low-level...

> Another consideration: what about bitbanged implementations of common peripherals like UART?

The machine module already has provision for software implementations (I2C and SPI use it), so a software UART would follow the same idea and be in the machine module.

aykevl, 2017-11-22:
Thank you for the feedback!

> I have in the past tried to invent a common interface/function for all these protocols but didn't succeed.

It might be possible, but I see a few more problems: performance (even if it's possible), memory requirements for the pulse train, and [extensibility / support for other protocols](https://blog.codinghorror.com/rule-of-three/).

I [don't really care about the name](http://bikeshed.com/), as long as it's clear. `lldrivers` is okay. Maybe `bitbang` as that's what we're doing here. I would suggest putting a `_` in front, however, just like `_thread`, `_io`, `_socket`, etc (see [PEP 8](https://www.python.org/dev/peps/pep-0008/#package-and-module-names)). These libraries are low-level wrappers around C modules.

Do you have a suggestion for how to proceed? I can make a pull request for the ESP8266 port, or simply go ahead and implement this module (however it's named) for the nRF51822.

dpgeorge, 2017-11-23:
> I would suggest putting a _ in front, however, just like _thread, _io, _socket, etc (see PEP 8)

uPy doesn't use a leading underscore much.  In `_thread` it's used to mirror CPython, and the other two cases currently in use are `_webrepl` and `_onewire` which are indeed low-level wrappers around C code.  `bitbang` may not be a good idea because that gives the intention of it being bit banged, whereas it may be that some ports implement the functions using other methods (eg hardware SPI).    The module is more of a place to collect implementations of very specific protocols that are timing sensitive, so maybe `_timingproto` is more descriptive, or even `llprotocol`.

In regards to the ws2812 driver: it would be a good chance to make the timing parameter more general like be an actual frequency (so you'd pass either 400 or 800, but any other value would also be allowed).  And some ports may provide an asynchronous ws2812 driver (eg CircuitPython has it) so might be worthwhile to keep that in mind as a future extension.

@tannewt how are you handling these low-level timing routines (ws2812, dht, onewire), are they all lumped together in one module?

> Do you have a suggestion for how to proceed?

Let's continue the discussion a bit.  The solution should work the same for stm32 and esp32 so think about how those ports would implement the idea, preferably using the same code as esp8266 and nrf.

aykevl, 2017-11-23:
> `bitbang` may not be a good idea...

Agreed.
`_timingproto` sounds a bit like a time protocol to me (NTP or machine.RTC), but I see your point. And `llprotocol` sounds a bit vague to me.

> In regards to the ws2812 driver: it would be a good chance to make the timing parameter more general like be an actual frequency (so you'd pass either 400 or 800, but any other value would also be allowed).

It seems like the 400KHz protocol isn't used anymore. I can't find anyone selling it that is not [explicitly deprecated](https://www.adafruit.com/product/1060), so I think we might be able to drop it. It makes the implementation a lot simpler and thus smaller/faster. Note that only the WS2811 can be configured in the slow protocol, all newer chips (WS2812, WS2812B, SK6812) all use the fast (800KHz) protocol only. Also, CircuitPython does not support the 400KHz protocol at all.

> And some ports may provide an asynchronous ws2812 driver (eg CircuitPython has it) so might be worthwhile to keep that in mind as a future extension.

I've looked but can't find an asynchronous implementation - do you mean [DotStar](http://circuitpython.readthedocs.io/projects/dotstar/en/latest/api.html)? I'm interested in knowing how it's done.

For reference:
http://circuitpython.readthedocs.io/en/1.x/shared-bindings/neopixel_write/__init__.html

tannewt, 2017-11-23:
@dpgeorge ws2812, dht, onewire are all handled by separate low level pieces.

* ws2812 (aka neopixel) is handled by [`neopixel_write.neopixel_write`](https://circuitpython.readthedocs.io/en/2.x/shared-bindings/neopixel_write/__init__.html) just like micropython's esp.neopixel_write IIRC.
* dht is handled by [`pulsio.PulseIn`](https://circuitpython.readthedocs.io/en/2.x/shared-bindings/pulseio/PulseIn.html#pulseio.PulseIn.resume) which can also handle IR reception.
* onewire bit reading and writing is handled by [`busio.OneWire`](https://circuitpython.readthedocs.io/en/2.x/shared-bindings/busio/OneWire.html) but the protocol level is a separate Python library [we're working on](https://github.com/adafruit/circuitpython/issues/338).

In general, we try to limit the C code to the timing critical pieces to save code size.

NeoPixel is currently synchronous for us though we may switch to doing it with DMA in the future. If we do then we'll likely leave the API the same and block in the off chance we get called a second time really quickly.

As @dpgeorge points out, we do DotStar with SPI. (Its also not timing critical because of the clock line.)