micropython/micropython #3455: mkrules.mk: the target for pre-processed files never runs. Ever.

adritium, 2017-11-23:
The target for pre-processed files is:
```
vpath %.c . $(TOP)
$(BUILD)/%.pp: %.c
	$(ECHO) "PreProcess $<"
	$(Q)$(CC) $(CFLAGS) -E -Wp,-C,-dD,-dI -o $@ $<
```
Unfortunately, it never runs! This is because `$(BUILD)/%.pp` is not a prerequisite of any target.

To get it to run, you could add it as a prereq of the object file though you also need to add `.PRECIOUS: $(BUILD)/%.pp` otherwise gnu make will automatically delete the `.pp` files (see [reason why](https://www.gnu.org/software/make/manual/html_node/Chained-Rules.html))

```
.PRECIOUS: $(BUILD)/%.pp

vpath %.c . $(TOP)
$(BUILD)/%.o: %.c $(BUILD)/%.pp
	$(call compile_c)
```

Of course you could argue that since the compile statement does not actually use `.pp` file in the process of creating the `.o`, it's not really a prerequisite . . . and I think that's a valid point. My solution above merely tacks on the dependency to force the recipe for `.pp` to execute.

However, you could modify the compile command so that `.pp` is an actual prerequisite while at the same time reducing compilation time (if you want the preprocessor output in the first place): use the `.pp` as an input to the next stage of compilation.

`gcc -c -x cpp-output -o main.o main.pp`

The `-x cpp-output` says **treat the input as the output of the preprocessor**

Should the `.pp` generation be optional? There's no way to output `.pp` and compile with one invocation. If you don't want the `.pp`, yeah you're saving time if you only invoke `gcc` once but it's gotta be miniscule; the preprocessing gets performed anyway so it's only the overhead of invoking `gcc` twice that you're saving.

dpgeorge, 2017-11-23:
> The target for pre-processed files is: ... Unfortunately, it never runs!

It's not supposed to run for normal builds.  As far as I remember, this rule was added so that you can generate preprocessed files on demand, if needed, for example so you can see how macros are expanded to debug them.  You can force a build of a .pp by: `make build/py/vm.pp`

adritium, 2017-11-23:
I like generating them all the time. There should be an option to pass in `MAKE_PP=1` like for `CROSS=1`

dpgeorge, 2017-11-24:
> I like generating them all the time.

It should be possible to do this by making your own local `makefile`/`GNUmakefile` (or failing that, by applying a simple patch to the build files in your local copy of the code).

> There should be an option to pass in MAKE_PP=1 like for CROSS=1

The build system is already complicated enough so I'd rather not add more options.

closed 2017-11-24 by adritium