micropython/micropython #3456: Add .exe suffix to PROG=mpy-cross if on windows

adritium, 2017-11-24:
https://github.com/micropython/micropython/blob/48f6990fbc54ac29b15fa765d431244d1dcb5f21/mpy-cross/Makefile#L17

`PROG=mpy-cross` results in following link step: `gcc -o mpy-cross build/py/mpstate.o ...` but the output executable still has `.exe` suffix. Consequently, this line fails when run on Windows

https://github.com/micropython/micropython/blob/48f6990fbc54ac29b15fa765d431244d1dcb5f21/py/mkrules.mk#L136

```
strip  mpy-cross
strip: 'mpy-cross': No such file
../py/mkrules.mk:134: recipe for target 'mpy-cross' failed
make: *** [mpy-cross] Error 1
```

We should add a check for whether we're on Windows and if so, `PROG=mpy-cross.exe`

adritium, 2017-11-24:
It turns out that on Windows, if you set `PROG=mpy-cross.` (with a dot) it will output it as `mpy-cross` (without the dot). I think it'd be straightforward to `PROG=mpy-cross.exe` if we're on windows

chrismas9, 2017-11-24:
Unfortunately it's a bit more complicated than that. Make clean tries to delete mpy-cross.exe.map and I think there are other problems as well. I had a solution in PR #3310 but it broke the Windows port which also uses PROG= but has to compile under Windows and cross compile under Linux. A full solution needs to know whether it is compiling for Windows (eg Windows port) or for the host (eg mpy-cross).

I have been out of action for a while, but I will revisit #3310 soon with a complete fix for all Windows targets. Then I want to get back to #3340 to allow Windows to interpret symlinks used for modules.

adritium, 2017-11-30:
@chrismas9  so this is a duplicate?

chrismas9, 2017-11-30:
@adritium. Yes, it's a duplicate of #3310 and #3370. I'm working on #3370 now which will combine the mpy-cross fix from #3310 (which was closed) with the Windows port fix I started in #3370. I'm testing now, still have to update documentation.

closed 2017-11-30 by adritium