micropython/micropython #3462: Facilitate micropython execution in a cooperative multitasking OS

adritium, 2017-11-30:
My OS is cooperative multitasking i.e. the context switch must be explicitly initiated via `RTOS.yield()` which means that in order for me to be able to integrate uPython into my RTOS, I have to find a spot to place `RTOS.yield()` call and I've found it.

Original code, **vm.c**

```
    #define DISPATCH() do { \
        TRACE(ip); \
        MARK_EXC_IP_GLOBAL(); \
        goto *entry_table[*ip++]; \
    } while (0)
```

Modified code:

```
    #define DISPATCH() do { \
        TRACE(ip); \
        MARK_EXC_IP_GLOBAL(); \
        RTOS.yield();\
        goto *entry_table[*ip++]; \
    } while (0)

```

But this is not efficient because it yields after every-single opcode and I don't need it to break after every single opcode.

The next optimization step is to say **only yield every X opcodes**

```
    #define DISPATCH() do { \
        TRACE(ip); \
        MARK_EXC_IP_GLOBAL(); \
        if(counter++ % NUM_OPS_EXC == 0) RTOS.yield();\
        goto *entry_table[*ip++]; \
    } while (0)

```

But I don't really care how many opcodes are executed, I care how much time is taken i.e. 5us, 50us, etc so what I'd really like to do is to have a downcounter that is decremented by an ISR and saturates (doesn't wrap around).

```
    #define DISPATCH() do { \
        TRACE(ip); \
        MARK_EXC_IP_GLOBAL(); \
        if(DownCounter == 0) RTOS.yield();\
        goto *entry_table[*ip++]; \
    } while (0)
```
In this config, total execution time is **DownCounter timeout** + **longest opcode execution time**

Since `#define DISPATCH()` already exists, why even create this issue? In a future refactoring of uPython, `#define DISPATCH()` may be removed and I'll have to find another point to "hack-in" my coop-multitasking fix and it may not be as clean and simple as today's. 

I request a placeholder (construct) - macro, function callback, etc - where user-defined `RTOS.yield()` can be inserted.

Given today's layout, the following makes sense to me:

```
   #define RTOS_YIELD()\
      if(counter++ % NUM_OPS_EXC == 0) RTOS.yield();\

    #define DISPATCH() do { \
        TRACE(ip); \
        MARK_EXC_IP_GLOBAL(); \
        RTOS_YIELD()\
        goto *entry_table[*ip++]; \
    } while (0)
```

dpgeorge, 2017-12-01:
There are actually already hook macros to accomplish what you are asking for:
```
MICROPY_VM_HOOK_INIT
MICROPY_VM_HOOK_LOOP
MICROPY_VM_HOOK_RETURN
```
For efficiency these hooks are not executed every bytecode but rather only at "jumps" and returns.  The esp8266 port uses them because it is essentially a cooperative multitasking system (needed for the wifi to do its processing).  It uses a divisor as well.  See ports/esp8266/mpconfigport.h for hints how to do that.

You might also be able to do it a different, more efficient way: enable MICROPY_ENABLE_SCHEDULER and then use a timer IRQ to call `mp_sched_schedule(callback, arg)`, with the callback being a function that calls `RTOS.yield()`.  That way the yield will be executed from the main task (ie not from within the IRQ) and it will always be executed between bytecodes, and Python operations (including memory allocation) will be atomic with respect to the yield.

adritium, 2017-12-01:
> ...only at "jumps" and returns

That's not deterministic enough. What if the execution of the `.py` goes for > my watchdog period? The authors of the `.py`s will not be the same as the authors of the RTOS.

> See ports/esp8266/mpconfigport.h for hints how to do that.

It's pretty much what I wanted to do: count to a certain number then call a function.

> For efficiency...

In an RTOS environment, determinism trumps efficiency; I can't have the micropython thread/task causing a watchdog timeout ever.

In order to accomplish determinism, there needs to be a `MICROPY_VM_HOOK_OPCODE` added to `#define DISPATCH()` so that if an application requires tight timing, there's the option to check after every opcode and if it's not used, it won't be compiled in.

stinos, 2017-12-03:
> That's not deterministic enough. What if the execution of the .py goes for > my watchdog period?

Just wondering, but how sure are you that yielding in DISPATCH is determinstic enough? There's still functions being called in between so unless you control all of them and depending on what OS it is you are targetting you're going to have a pretty hard time to get your 5uSec.

robert-hh, 2017-12-03:
Not to mention the 100µs++ execution jitter of the ESPxxx platforms.

dpgeorge, 2017-12-03:
> In order to accomplish determinism, there needs to be a MICROPY_VM_HOOK_OPCODE added to #define DISPATCH()

As @stinos mentions, even doing this won't be enough to get each opcode executing within strict time limits.  For example sorting a list is a single opcode (the function/method call) and that can take an arbitrary time depending on the size of the list (eg on pyboard sorting a sorted list of 100 elements takes 2.6ms, and 1000 elements takes 249ms).  Also inserting into a dict may require a rehash of the dict and then a garbage collection which can make the insertion time vary from between 30us to 1500us (approx).

adritium, 2017-12-03:
Assuming I turned off garbage collection and left it as the user's responsibility to call, would I still have a problem with 

> sorted list of 100 elements takes 2.6ms

dpgeorge, 2017-12-03:
> Assuming I turned off garbage collection and left it as the user's responsibility to call, would I still have a problem with
> 
>     sorted list of 100 elements takes 2.6ms
> 
> ?

Yes.  And there are probably more examples like that.  But note that it's not a Python/MicroPython specific issue: even if you were programming in C you still couldn't sort a (large) list and have fast, deterministic cooperative thread switching.  You'd need to recode the sorting function to "yield" at certain places during the sort.

If you need to offer a deterministic environment with *cooperative* yielding then you'll need to either 1) go through each MicroPython operation (eg method calling, for loop, list.sort, dict lookup, str.find, etc, there are heaps of them) and characterise its timing and then write a guide to tell which operations can be used and which can't; 2) insert `rtos_yield()` in many places throughout the code base to guarantee that it'll be able to yield within a given time frame.

adritium, 2017-12-04:
Yep I see. I guess if I want really want determinism, I'll have to go the asynchronous-timer route.

Thanks everyone!

adritium, 2017-12-07:
I re-thought this: while asynchronous timer is the ideal solution . . . it may not be the most feasible solution. I'm using a cooperative RTOS which I'd have to majorly overall to give it some preemptive abilities.

And if all this can be avoided by modifying `#define DISPATCH()` to add _one macro_ like so:

```
#define DISPATCH() do { \
        TRACE(ip); \
        MARK_EXC_IP_GLOBAL(); \
        RTOS_YIELD();\
        goto *entry_table[*ip++]; \
    } while (0)
```

I understand that this solution would not solve the problem of C-functions that are unbounded; I would still need to replace them or manually instrument them with my `RTOS.yield()` but those would be the exceptions. Adding `#define RTOS_YIELD()` gives the user more control in the execution of the VM in the form of granularity in time. If they don't use it, it'll be compiled out just like the dozens of other macros.

int19h, 2019-07-26:
Not directly related to the original scenario, but having such a hook would be a huge step towards enabling debugging directly in MicroPython, and could be the first step towards solving e.g. #3009. With such a hook, it would be possible to implement line-level tracing on top of it, and that in turn could provide `sys.settrace` - or better yet, used directly to implement a small and frugal debugger that is a subset of full-fledged `pdb`.