micropython/micropython #3463: persistentcode.c: save_raw_code() remove unnecessary tail recursion

adritium, 2017-12-02:
https://github.com/micropython/micropython/blob/bb047558da6d7e1857e1d4c79547776976353cd3/py/persistentcode.c#L351

Having recursion in embedded environment makes it impossible to allocate memory in linker file: unless max recursive depth can be computed at compile time, you’ll either allocate too much or too little stack space. And if there is a stackoverflow, an OS without virtual memory can have other threads’ state corrupted because the entire memory map is available from any running thread. And no, MPU will not fix this: if you don’t know how big the stack can be then you can’t set the MPU.

Furthermore, in this case there’s no reason for the recursion other than to convey the idea that each subsequent entry in `const_table` is a `mp_raw_code_t`. And if that is the intent, that can easily be accomplished with a couple lines of a comment.

dpgeorge, 2017-12-03:
This case in persistentcode.c is not pure tail recursion because there is a loop around the recursive call to save_raw_code.  I don't know how to eliminate such recursion without dynamically allocating some memory to store the recursive state of the function.  Did you have any ideas how to eliminate the recursion?

adritium, 2017-12-03:
So I'm clear, is the state that's needed to be saved the value of `const_table`?

dpgeorge, 2017-12-03:
> is the state that's needed to be saved the value of const_table?

You'll need to save all state/variables associated with that loop: `i`, `rc->data.u_byte.n_raw_code`, `print`, `const_table`.  From a deterministic point of view it's usually favourable to use C recursion rather than heap allocation.

adritium, 2017-12-04:
I see your point. I realize that you'd have to use a user stack to emulate the C-stack.

But, if you used the heap to build up the C-call stack manually, at least you wouldn't overflow your heap because you'd get an allocation error from within uPy's heap manager so it'd be safer.

Also - and I'm not a c-stack expert - would you actually use more RAM by saving the loop state vs how much a stack-frame takes up?

I agree that time-wise, a recursive call is preferred to a heap allocation which can trigger all sorts of operations but in this case, I think I'd trade-off a little more time for safety.

So, I agree that it doesn't make sense for YOU to spend your time on this but if I submitted a version of `save_raw_code` (which is macro-disabled by default) and uses user-stack vs recursion, would you accept it to be merged into the code?

dpgeorge, 2017-12-04:
> at least you wouldn't overflow your heap because you'd get an allocation error from within uPy's heap manager so it'd be safer.

There is `MP_STACK_CHECK()` to check that the C stack doesn't overflow, which could be added in this case.

> would you actually use more RAM by saving the loop state vs how much a stack-frame takes up?

It really depends.  And it depends as well on the architecture (eg x86, ARM).

> if I submitted a version of save_raw_code (which is macro-disabled by default) and uses user-stack vs recursion, would you accept it to be merged into the code?

I would suggest you spend your efforts elsewhere, eliminating recursion in other places.  The `save_raw_code()` function can run on embedded systems but usually it doesn't, it's usually just used for `mpy-cross` which is run on a (non-embedded) host system.

If you run a static analyser which shows the call-graph flow then you'll notice there is a fair bit of recursion in the main VM/runtime which is more critical to "remove" than that in `save_raw_code()`.  Eg it would be nice to have a non-recursive list sort routine.

(Note that the config option `MICROPY_MULTIPLE_INHERITANCE` was added to help eliminate recursion when that is an important factor.)

adritium, 2017-12-04:
I actually will need to use `save_raw_code` but if there's a bunch of recursion in the VM runtime, I'd agree it's much more critical to remove recursion in the runtime than in something that barely runs during my module's lifetime.

Thanks for the feedback.

closed 2017-12-04 by adritium

adritium, 2017-12-04:
@dpgeorge in what other locations is recursion present? (I'm in the middle of getting `cflow` to work)

dpgeorge, 2017-12-05:
> in what other locations is recursion present?

mp_execute_bytecode is (indirectly) recursive.  But otherwise, please run a tool to find out the details.