micropython/micropython #3465: 1.9.3 -include $(OBJ:.o=.P) not working

adritium, 2017-12-03:
I'm launching `make` in `micropython-1.9.3\ports\stm32` with command

`make all --debug=bvjm --trace V=1 LC_ALL=C BOARD=PYBV11 MICROPY_FLOAT_IMPL=double V=1`

An entire line in the build logging:

`../../py/mkrules.mk:47: update target 'build-PYBV11/py/mpstate.o' due to: ../../py/mpstate.c`

but in 1.9.2

`../py/mkrules.mk:47: update target 'mkfile-build/py/mpstate.o' due to: ../py/mpstate.c mkfile-build/genhdr/qstrdefs.generated.h ../py/mpstate.h ...` (keeps going)

I'm using same version of `make` and `mkrules.mk`, `py.mk`, and `mkenv.mk` are trivially different from 1.9.2 to 1.9.3 so I think it's an issue with `micropython-1.9.3\ports\stm32\Makefile`

dpgeorge, 2017-12-04:
It's difficult to see what's being reported here as an issue/bug.  Using unmodified master, `make BOARD=PYBV11 MICROPY_FLOAT_IMPL=double` works fine.  Please try doing `make BOARD=PYBV11 clean` first and then rebuild.  If there's still an issue please show how to reproduce it.

adritium, 2017-12-04:
The issue is that the `.P` file is not being imported so the only dependencies listed are the `.c` file for the `.o`; I will unzip a new 1.9.3 and try it again.

adritium, 2017-12-04:
I re-ran it and the same issue: instead of listing all the dependencies in the corresponding `.P` file i.e. `mpstate.p` 

```
build-PYBV11/py/mpstate.o: ../../py/mpstate.c ../../py/mpstate.h \
 c:\program\ files\ (x86)\codesourcery\sourcery_codebench_lite_for_arm_eabi\bin\../lib/gcc/arm-none-eabi/4.7.2/include/stdint.h \
 c:\program\ files\ (x86)\codesourcery\sourcery_codebench_lite_for_arm_eabi\bin\../lib/gcc/arm-none-eabi/4.7.2/../../../../arm-none-eabi/include/stdint.h \
 ../../py/mpconfig.h mpconfigport.h boards/PYBV11/mpconfigboard.h \
 ../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal.h \
 boards/PYBV11/stm32f4xx_hal_conf.h \
 ../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_rcc.h \
 ../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_def.h \
 ../../lib/stm32lib/CMSIS/STM32F4xx/Include/stm32f4xx.h \
 ../../lib/stm32lib/CMSIS/STM32F4xx/Include/stm32f405xx.h \
 ../../lib/cmsis/inc/core_cm4.h ../../lib/cmsis/inc/core_cmInstr.h \
 ../../lib/cmsis/inc/cmsis_gcc.h ../../lib/cmsis/inc/core_cmFunc.h \
 ../../lib/cmsis/inc/core_cmSimd.h \
 ../../lib/stm32lib/CMSIS/STM32F4xx/Include/system_stm32f4xx.h \
 ../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc/Legacy/stm32_hal_legacy.h \
 ../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_rcc_ex.h \
 ../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_gpio.h \
 ../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_gpio_ex.h \
 ../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_dma.h \
 ../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_dma_ex.h \
 ../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_cortex.h \
 ../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_adc.h \
...
```

only the `.c` is listed i.e.

```
      Must remake target 'build-PYBV11/py/mpstate.o'.
../../py/mkrules.mk:47: update target 'build-PYBV11/py/mpstate.o' due to: ../../py/mpstate.c
echo "CC ../../py/mpstate.c"
CC ../../py/mpstate.c
```

adritium, 2017-12-04:
At the bottom of `mkrules` is the line that imports the `.P` into the build process

`-include $(OBJ:.o=.P)`

The three `*.mk`s are trivially different so the issue must be with `micropython-1.9.3\ports\stm32\Makefile`

dhylands, 2017-12-04:
Note that when doing a clean build there will be no .P files so there won't be any dependencies. This is fine as the .o files won't exist, and the fact that the .o only depends on the .c is enough to trigger it to compile.

This is why there is a -include (the minus tells make to ignore missing .P files).

The .P files will be created/updated each time the compiler runs, and will only be used on the second invocation of make after a clean.

adritium, 2017-12-06:
I ran it a few times and you are correct and it makes sense now.

If *.o doesn't exist, doesn't make sense to ask "which headers changed since last build" because there is no last build.

If *.o DOES exist, THEN it make sense to look at the headers (and the .c of course) if any of them changed.

closed 2017-12-06 by adritium