micropython/micropython #3466: Port tools\make-frozen.py to C and merge into micropython C-code

adritium, 2017-12-04:
I have the following workflow that I need to accomplish:

1. Take raw `.py` and send to my microcontroller (via SPI, USB, WiFi, etc)
2.  Save `.py` to flash (I have no file-system)
3.  Use on-board compiler to compile it 
4.  Run it

I have discovered that as of 1.9.3, there is no way to do this.

If I had a file-system so that I could use the POSIX function `open`, the following sequence of calls would be available to me to accomplish my task:

pyexec.c: `pyexec_file("myScript.py`) 
-> `parse_compile_execute(filename, MP_PARSE_FILE_INPUT, EXEC_FLAG_SOURCE_IS_FILENAME)`
->  `lex = mp_lexer_new_from_file(source);`
->  lexer.c `mp_lexer_new_from_file(const char *filename)`
... then continuing in **pyexec.c: parse_compile_execute()**

But because I don't have a file-system, I can't use `mp_lexer_new_from_file`.

The other way to "import" raw `.py` is to build it in with the firmware via the `make` variable `FROZEN_DIR`
https://github.com/micropython/micropython/blob/155ec21e4933f75ee98bbd3e125e4304fb188170/py/mkrules.mk#L98

However, that only works during the build process which calls `tools\make-frozen.py`. Therefore, the only way for me to accomplish this is to port `tools\make-frozen.py` to C.

If this issue _sounds_ like a duplicate . . . that's because it's actually a **subset** of #2709 "_Compile Python Scripts On the Go_" but it's still a valid **standalone** issue.

In #2709 the goal is to be able to load in a `.py` **within the micropython virtual machine** and compile it (and save it). Well, to accomplish that, you first need a C-API to do that.

adritium, 2017-12-04:
Yes I know I could `mpy-cross` the `.py`s but that leads to possible compatibility issues: `mpy-cross` would have to be bundled into our PC app and if the PC `.mpy` version is different then our module firmware would have to be reflashed . . . easier to simply do the compilation on the module itself.

adritium, 2017-12-04:
Looks like this won't be hard at all.

Here's a sample `.py` 

```
import pyb

# print("Executing main.py")

ledObj = pyb.LED(1)

while True:
	ledObj.toggle()
	ledObj.toggle()
```

And here's the resulting `frozen.c`

```
#include <stdint.h>
const char mp_frozen_str_names[] = {
"main.py\0"
"\0"};
const uint32_t mp_frozen_str_sizes[] = {
111,
};
const char mp_frozen_str_content[] = {
"import pyb\n\n# print(\"Executing main.py\")\n\n\nledObj = pyb.LED(1)\n\nwhile True:\n\x09" "ledObj.toggle()\n\x09" "ledObj.toggle()\n\n\0"
};
```

The only thing for me to figure out is what those hex characters are.

robert-hh, 2017-12-04:
\x09 is the tab character, \0 is the null-terminator of a C-string, \n is Newline (\x0a)

peterhinch, 2017-12-05:
The OP indicates that step 2 (saving to flash) is not required and you can use USB for connectivity. Unless I'm missing something your workflow describes the way test scripts are run on the Pyboard. In the source tree see the `tests` directory, also the utility `tools/pyboard.py`.

adritium, 2017-12-05:
I tried to remove unnecessary details but looks like I removed too much:
* I will be saving the raw `.py` to flash so that when I reboot, the file won’t be lost. However, that won’t change the fundamental problem because from the perspective of any potential solution, reading code from RAM vs flash is equivalent. 
* There is no REPL connection to the PC although REPL mode for code execution will be available.

adritium, 2017-12-05:
If I were to feed in the code from the raw `.py` to REPL mode, roughly how much more RAM would that take vs freezing the code in the first place a la `frozen.c`?

If you look at `frozen.c`, it’s trivial to produce it “dynamically”.

adritium, 2017-12-06:
I was `missing the forest for the trees`. I realized that you do not need to call `lib\utils\pyexec.c: pyexec_frozen_module()` because that function (and file) is full of helper functions i.e. there's no rule that says you MUST call `pyexec_frozen_module()`.

Assuming I have the following vars defined somewhere

```
char * name; // name of my .py file
uint32_t size; // size in bytes of .py file
char * py_file_content; // .py file text contents
```

I can accomplish this with the following sequence of calls:

```
qstr source = qstr_from_strn(name, name_len);
mp_lexer_t *lex = MICROPY_MODULE_FROZEN_LEXER(source, py_file_content, len, 0);

mp_parse_tree_t parse_tree = mp_parse(lex, input_kind);
module_fun = mp_compile(&parse_tree, source_name, MP_EMIT_OPT_NONE, exec_flags & EXEC_FLAG_IS_REPL);
mp_call_function_0(module_fun);
```

adritium, 2017-12-06:
In this issue, I was trying to figure out how to load in `.py` file and compile and run it and I realized, the API already supports it but it's not clear enough.

`mp_find_frozen_module` returns a lexer from  `.py` code from _from a specific location_ i.e. the files represented by the variables **frozen.c**. 

I think we should have an API that can return a lexer of a file located anywhere **and** `mp_find_frozen_module` should call this new API.
```
typedef struct ram_flash_file_
{
    char * name;
    uint32_t nameSize;
    char * content;
    uint32_t contentSize;
} ram_flash_file_t;

int main()
{
void *data;

ram_flash_file_t py_file;
py_file.name = getFileName();
py_file.name_size = getFileNameSize();
py_file.content= getFileContent();
py_file.contentSize = getFileSize();

mp_get_data(py_file, &data);

parse_compile_execute(data, MP_PARSE_FILE_INPUT, 0);
}

void mp_get_data_mem_file(ram_flash_file_t  py_file, void ** data)
{
    qstr source = qstr_from_strn(py_file.name, py_file.nameSize);
    mp_lexer_t *lex = MICROPY_MODULE_FROZEN_LEXER(source, py_file.content, py_file.contentSize, 0);
    
    *data = lex;
}
```

closed 2018-05-02 by adritium