micropython/micropython #3469: 68c28174 breaks utf8 decode with ignore

ryannathans, 2017-12-05:
b'\xff\xfe'.decode('utf8', 'ignore')

used to work but now produces UnicodeError exception due to commit https://github.com/micropython/micropython/commit/68c28174d0e0ec3f6b1461aea3a0b6a1b84610bb

On CPython this still works as expected:

>>> b'\xff\xfe'.decode('utf8', 'ignore')
''

dpgeorge, 2017-12-07:
> used to work

Going back to the commit before UTF-8 checking was introduced, it seems that "ignore" didn't work as CPython does it.  I got the behaviour (unix version):
```python
>>> b'a\xff\xfea'.decode('utf8', 'ignore')
'a\x00\x00a'
```
That's not any kind of correct behaviour.

@ryannathans in what way did you originally use "ignore" and how was it working originally for you?

ryannathans, 2017-12-11:
@dpgeorge the function was used to decode utf8 text out from the S-RAM without throwing exceptions.

For example, I stored crash dumps in the backup S-RAM in under 1024 bytes, rebooted, then read these out and 'decoded' them on next boot. The exception text would be trailed by \x00 (because \xff parsed as \x00 for some reason) so I just stripped off all trailing \x00.

dpgeorge, 2017-12-13:
> I stored crash dumps in the backup S-RAM in under 1024 bytes, rebooted, then read these out and 'decoded' them on next boot. The exception text would be trailed by \x00 (because \xff parsed as \x00 for some reason) so I just deleted all \x00 bytes.

For this example you can use `rstrip(b'\xff')` to remove trailing bytes before converting to a str.

> It was also used for reading strings from an AT modem over UART which sometimes didn't have correct utf8 characters in it. I just accepted the behavior of \x00 (again deleting all \x00 bytes), without realising that CPython output '' instead of \x00 for each invalid character.

In this case you'll need to write some custom utf8 validation code, at least until uPy has support to properly ignore invalid utf8 sequences.

jonnor, 2024-09-29:
Same behavior on MicroPython v1.24.0-preview.276.g1897fe6227 (Unix port).

github0null, 2024-10-10:
> > 曾经工作
> 
> 回到引入 UTF-8 检查之前的提交，似乎 “ignore” 不像 CPython 那样起作用。我得到了行为（unix 版本）：
> 
> ```python
> >>> b'a\xff\xfea'.decode('utf8', 'ignore')
> 'a\x00\x00a'
> ```
> 
> 这不是任何正确的行为。
> 
> 您最初以何种方式使用 “ignore”，它最初对您来说是怎样的？

Same problem here, i am writing a uart decoder, sometimes garbled characters appear and I need to ignore them, but `decode('utf8', 'ignore')` will throw a error.

The expected behavior should look like this:

[image]

```python
# 1. throw Error
print((b"\xAA\x80 abc").decode("utf-8","strict"))
# ---------------------------------------------------------------------------
# UnicodeDecodeError                        Traceback (most recent call last)
# Cell In[17], line 2
#       1 # 1. throw Error
# ----> 2 print((b"\xAA\x80 abc").decode("utf-8","strict"))

# UnicodeDecodeError: 'utf-8' codec can't decode byte 0xaa in position 0: invalid start byte

# 2. replace to \uFFFD reference: https://www.fileformat.info/info/unicode/char/fffd/index.htm
b = b"\xAA\x80 abc"
print(b.decode("utf-8","replace"))
print(' -> after replaced: ', b.decode("utf-8","replace").encode())
# �� abc
#  -> after replaced:  b'\xef\xbf\xbd\xef\xbf\xbd abc'

# 3. Add backslash
print((b"\xAA\x80 abc").decode("utf-8","backslashreplace"))
# \xaa\x80 abc

# 4. ignore it from input
print((b"\xAA\x80 abc").decode("utf-8","ignore"))
# abc
```

github0null, 2024-10-11:
> Same problem here, i am writing a uart decoder, sometimes garbled characters appear and I need to ignore them, but `decode('utf8', 'ignore')` will throw an error.
> 
> The expected behavior should look like this:

A temporary fix:

Use `predecode(...)` to replace or ignore invalid code before call `bytes.decode(...)`

[image]

Source code of function `predecode`:

https://gist.github.com/github0null/4626ee402eb1816dc75c03f480bdebf8

github0null, 2024-10-11:
> Same problem here, i am writing a uart decoder, sometimes garbled characters appear and I need to ignore them, but `decode('utf8', 'ignore')` will throw an error.
> 
> The expected behavior should look like this:
> 

A temporary fix:

https://gist.github.com/github0null/4626ee402eb1816dc75c03f480bdebf8

Use `predecode(...)` before call `bytes.decode(...)`

[image]