micropython/micropython #3472: Transition to compatibility with C++ compiler

adritium, 2017-12-06:
Some people have a C++ codebase and it'd be preferable to compile everything with C++ compiler.
Also, if the code were compile-able with C++, you could incorporate it into C++ classes much easier.

I tried to "fix" the code so that it could compile with C++ but it turned out to be impossible.
The biggest issue was using C-array initialization constructs that are simply not allowed in C++.

For example, [see this stackoverflow question](https://stackoverflow.com/questions/47076204/c-how-to-assign-static-array-to-a-pointer-inside-a-struct) 

This is allowed in C but not in C++.
```
typedef struct _type_t
{
    int a;
    int b;
    int c[];
}type_t;

type_t var = {1,2,{1,2,3}};

// from the following
#define DEF_RULE_NC(rule, kind, ...) static const rule_t rule_##rule = { RULE_##rule, kind, #rule, { __VA_ARGS__ } };
```

Also, [designated initializers](https://stackoverflow.com/questions/18731707/why-does-c11-not-support-designated-initializer-lists-as-c99)

```
struct Person
{
    int height;
    int weight;
    int age;
};

int main()
{
    Person p { .age = 18 };
}
```

@dpgeorge  ?

stinos, 2017-12-06:
In my opinion there's no reason to compile everything as C++. For one, because it's C. Second: it it simply not needed. C++ consumes C APIs just fine, and uPy's works fine with C++.

adritium, 2017-12-06:
The problem is when you have to call C++ from micropython. How do you call a `Object.Function()` from C?

stinos, 2017-12-06:
You don't, or you create a C API for your C++ code. Which would only bee needed if you somehow need the uPy core itself to call C++. If you want to wrap C++ classes for use in uPy you can easily do so by compiling just files as C++

adritium, 2017-12-07:
Yes you can do all that but if micropython can be made compatible with C++ compiler at trivial cost, why not do it?

stinos, 2017-12-07:
I'd say because the contributers have better things to do, because there is no real need for it or at least no-one yest has shown a compelling use-case, and also I'm not sure it's that trivial.

adritium, 2017-12-07:
I'll agree with your first two points but why do you say it might not be that trivial?

stinos, 2017-12-07:
Just because I didn't check how much occurrences there are of the 2 cases you mention, nor whether there are other types of syntax which are valid in C but not in C++. For example it wouldn't surprise me if there are multiple places where implicit pointer (and possibly other) casts are used which are invalid in C++. And maybe some other things I'm not thinking of. And that would be at the lower warning levels. Crank up the warning level and you'd likely get signed <-> unsigned conversion warnings and so on. So yes these can all easily be rewritten without behaviour changes (I think), but all those combined could mean it's a pretty huge change and that alone makes it not super trivial. But again: I'm not sure, never tried it.

Apart from that: I'm fairly sure one of the reasons C was chosen is simply because there are more platforms supporting it. I don't know al possible compilers but it's not that long ago that I was working on some TI DSP devices and C++ support was terrible. So suppose we go like 'ok let's support C++' then build times should be multiplied by 2, because we'd wanna make sure it still works *both* for C and C++ compilers.

dpgeorge, 2017-12-08:
> In my opinion there's no reason to compile everything as C++. For one, because it's C. Second: it is simply not needed. C++ consumes C APIs just fine, and uPy's works fine with C++, from experience.

I agree.  uPy already plays relatively nicely within a C++ system.  For a concrete example see the micro:bit code at https://github.com/bbcmicrobit/micropython/ .  There the main parts of the system are C++ and uPy is compiled using gcc (not g++).

> The problem is when you have to call C++ from micropython. How do you call a Object.Function() from C?

You can write uPy functions/classes/modules in C++, and mix C++ with C.  See for example https://github.com/bbcmicrobit/micropython/blob/master/source/microbit/microbitdisplay.cpp .  Note that the file is a .cpp file but contains a definition of a uPy class.  Also note the big "extern C" that wraps most of the file, and the use of C++ code and calling of C++ methods from uPy.

> The biggest issue was using C-array initialization constructs that are simply not allowed in C++.

Yes, this is not allowed in C++ and is not easy to remove/replace with something else.

dpgeorge, 2017-12-08:
@stinos while we are on this topic of C++ compat: one thing I needed to modify in uPy for the micro:bit code to compile with g++ was obj.h.  g++ didn't like initialising unions with a named member (structs are ok).  So I added this code: https://github.com/bbcmicrobit/micropython/blob/master/inc/py/obj.h#L275-L299 .  Did you come across such issues?

adritium, 2017-12-08:
If you’re asking me, yes I did (using 1.9.2). 
By the way, what does the cast do here?

```
#define MP_DEFINE_CONST_FUN_OBJ_0(obj_name, fun_name) \
    const mp_obj_fun_builtin_t obj_name = \
        {{&mp_type_fun_builtin}, false, 0, 0, fun_name}
#define MP_DEFINE_CONST_FUN_OBJ_1(obj_name, fun_name) \
    const mp_obj_fun_builtin_t obj_name = \
        {{&mp_type_fun_builtin}, false, 1, 1, (mp_fun_0_t)fun_name}
#define MP_DEFINE_CONST_FUN_OBJ_2(obj_name, fun_name) \
    const mp_obj_fun_builtin_t obj_name = \
        {{&mp_type_fun_builtin}, false, 2, 2, (mp_fun_0_t)fun_name}
```

dpgeorge, 2017-12-08:
> By the way, what does the cast do here?

It just casts the function pointer to be of type `mp_fun_0_t` which is the type of the first element of the union in mp_obj_fun_builtin_fixed_t, and so C++ has no problem initialising that element.

adritium, 2017-12-08:
Then shouldn’t you add the cast to the first one in the list too?

dpgeorge, 2017-12-08:
> Then shouldn’t you add the cast to the first one in the list too?

No, because the first one (MP_DEFINE_CONST_FUN_OBJ_0) already has the argument of the correct type `mp_fun_0_t`.

stinos, 2017-12-08:
> Did you come across such issues?

No, with g++ nor msvc. I'm always including obj.h but I don't effectively use those macros so the compiler doesn't consider the code..

closed 2018-05-02 by adritium