micropython/micropython #3474: MicroPython and emscripten

LeaNumworks, 2017-12-07:
Hi,
I work on a graphic calculator for the company [Numworks](https://github.com/numworks/epsilon). We have a Python app that uses MicroPython  and ran into some issues when using emscripten to build our [online simulator](https://www.numworks.com/simulator/).
- MicroPython assumes the stack is descending, which is not the case for us : in py/stackctrl.c, line 40 (if you want, we would be happy to make this patch for you)
```
mp_uint_t mp_stack_usage(void) {
   // Assumes descending stack
    volatile int stack_dummy;
   return MP_STATE_THREAD(stack_top) - (char*)&stack_dummy;
}
```
and in most example gc_collect() we found, when computing the length passed as an argument for gc_collect_root.
- We had trouble with garbage collection. According to emscripten’s [portability guidelines](http://kripken.github.io/emscripten-site/docs/porting/guidelines/portability_guidelines.html),  
"Code that scans registers or the stack" will not compile because 

>a variable in a register or on the stack may be held in a JavaScript local variable (which cannot be scanned). 

We fixed that by putting the optimization flag to -O0 when compiling MicroPython for emscripten.

This could be interesting for issues #3313, #2618 , #1561, #888.

dpgeorge, 2017-12-08:
Hi @LeaNumworks, nice work with your simulator!

> MicroPython assumes the stack is descending, which is not the case on emscripten.
> We had trouble with garbage collection as explained in emscripten’s portability guidelines.

Myself and a colleague have also lately been working on an Emscripten port and came across these two issues as well.  The first is easy to fix and we can provide a compile-time option in uPy to select the stack direction.

We could not fix the second issue yet, with the GC.  We have a build of uPy that runs under node (eg `node micropython.js <script.py>`) and running this on the uPy test-suite leads to a fair few failures which are due to the GC not being able to trace root pointers.  Even compiling with -O0 does not fix the issue.  What was your approach to dealing with this problem?

> Would you be interested in some PRs to get a working Emscripten build?

I'm definitely interested in having an Emscripten port, but the GC issue is a real show stopper at this point.

LeaNumworks, 2017-12-08:
I agree that having a working GC is a priority.

We only really did three things:

- Fix the stack direction
- Get the root pointers in [gc_collect](https://github.com/numworks/epsilon/blob/1da26f4f65a69075dab2ba9c5ff1d12d5936ae6c/python/port/port.cpp#L116) using setjmp
- Compile uPy with -O0

We did not run the uPY test-suite on our port so we may indeed be missing something. That being said, it does a great job for our use-case and the few scripts we tried this on did not seem to be leaking memory.

kkimdev, 2017-12-14:
@LeaNumworks Would you mind sharing your branch?

In case fixing the garbage collector properly is a huge headache, how about having a naive implementation of https://github.com/micropython/micropython/blob/master/py/gc.c for Emscripten port for now and migrating to https://github.com/WebAssembly/design/issues/1079 in the future when it's ready?

kkimdev, 2017-12-14:
@LeaNumworks Oh, I guess it's https://github.com/numworks/epsilon/tree/master/python nvm

alanjds, 2017-12-15:
Nice!

I had put a bounty on https://github.com/micropython/micropython/issues/888#issuecomment-234735603 to get something like this done. Not that $100 USD could possibly cover the costs, but is more about an incentive to get it rolling.

Thanks very much for your effort.

kkimdev, 2017-12-19:
nvm I took some time to read the source code and have a better understanding of the structure now.  I'll ask if I have any further question thanks!

kkimdev, 2018-01-03:
You can try @LeaNumworks 's stand-alone micropython here https://kkimdev.github.io/epsilon/index.html

repo: https://github.com/kkimdev/epsilon
I just stripped off non-micropython dependencies from https://github.com/numworks/epsilon/tree/master/python , it didn't require much work actually.

alanjds, 2018-01-03:
181 KB download / 665 KB unpacked !

This is nice!

dpgeorge, 2019-03-13:
A port to javascript was added in 7d675f3a17b435c4dc7880ff597ab81274721ae9

geowor01, 2019-04-24:
@LeaNumworks I am currently working on running MicroPython for the micro:bit with emscripten and emterpretify. I have applied your GC fixes, i.e. the descending stack, using setjmp, and I know that optimisations are not applied. Despite this, the GC still sweeps what it shouldn't, specifically, the local code_state pointer in fun_bc_call appears to not be stored in the stack but stored somewhere in emterpreter's local javascript heaps.

Am I right in thinking that you also use emterpretify?
If so, is there any reason you can think of that this might happen for me but not for the Numworks simulator?
i.e. How do you handle emterpretify storing pointers in heaps?
Is that the same issue you are trying to prevent with -O0?

dpgeorge, 2019-04-26:
@geowor01 you may want to test with #4723 to see if it helps your case

geowor01, 2019-04-29:
Thanks @dpgeorge! That does appear to fix the problem.

jonnor, 2023-07-30:
This issue can probably be closed now? The JavaScript port mentioned above has been renamed to WebAssembly, and should be generally functional. Any problems with it should probably be handled in specific issues?

mattytrentini, 2023-07-30:
Appears resolved.

closed 2023-07-30 by mattytrentini